pi-hole/PADD #97: Version check fails.

deepcore2, 2020-04-20:
My PADD says it's out of date, but I run 3.1 which is latest ASFAK. I looked at the version check in the code and that seems to be faulty. TYhe check is:

curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]'

Which returns nothing. Location is not withj a captital L. C The following check returns 3.1:

curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]'

Please fix.

sgu222e, 2020-04-22:
Duplicate of #73

downtime0, 2020-05-22:
@deepcore2 thanks for case sensitivity fix. I fixed mine the same and later found someone suggest removing the "grep" and simplify the line.

<code>
padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | awk -F '/' '/location/ {print $NF}' | tr -d '\r\n[:alpha:]')
</code>

regnare, 2020-06-05:
> @deepcore2 thanks for case sensitivity fix. I fixed mine the same and later found someone suggest removing the "grep" and simplify the line.
> 
> ` padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | awk -F '/' '/location/ {print $NF}' | tr -d '\r\n[:alpha:]') `

You could also pass -i to grep to make it case insensitive.
```
# PADD version information...
padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep -i 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')
```

PR #111 (2020-07-12): Use GitHub API for remote version detection.
Use the GitHub API for remote version detection. 

Bumps to version v3.2.2.

Closes #73, closes #82, closes #90, closes #107 
Resolves #110, resolves #97, resolves #73, resolves #80, resolves #87
closed 2020-07-12 by dschaper