pi-hole/PADD #100: Newest Update Does Not Fit 3.5" Display Correctly

Twiggy159, 2020-05-08:
**Describe the bug**
Upon updating PADD to the latest version, it appears that everything above the STATS Header row (Including the header row itself) is no longer displaying.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to the latest PADD Version via the following command:
`cd ~`
`wget -N https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh`

**Expected behavior**
Everything should be displayed correctly on the display

**Screenshot(s)**
Below is a PuTTY session screenshot of the display, I've outlined in red what the actual display shows.
[image]

**Additional context**
Current settings from the `sudo dpkg-reconfigure console-setup` options:
- UTF-8
- Guess optimal character set
- Terminus
- 6x12 (framebuffer only)

I've also tried the 8x14 setting as well but it cropped even more off the screen.

MatKier, 2020-05-23:
Thanks for opening this issue. I just started using PADD and it was driving me mad why nothing made the "Regular" size fit on a 3.5 (480 x 320) screen.

The last Version to completely fit on a 3.5 screen with Terminus font at 8x14 is v3.0.2.

Additional line breaks after the bold headings ("PI-HOLE", "STATS", "NETWORK", "SYSTEM") seem to be the reason for this.

PADD V3.1:
[image]

PADD V3.0.2:
[image]

Goldfire86, 2020-05-24:
Is there a way to rollback to the previous version?

MatKier, 2020-05-24:
Just download the old script from the commit history:
https://raw.githubusercontent.com/jpmck/PADD/d1d6cec9b6cc8765dc64bb4bd624b5f7143cef80/padd.sh

Goldfire86, 2020-05-24:
> Just download the old script from the commit history:
> https://raw.githubusercontent.com/jpmck/PADD/d1d6cec9b6cc8765dc64bb4bd624b5f7143cef80/padd.sh

Thank you, appreciated.

Jarsky, 2020-05-26:
This is a known issue, there are several fixes in PADD 3.1  you should make. 

Fix the '====' bug for line spacing
Line 569 - Change
`CleanEcho() {  echo -e $1 "${ceol}" }`
to 
`CleanEcho() {  echo -e "${ceol}$1" }`

Fix router addresses in DHCP section for multiple routers
Line 331 - Change
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { print $3 }')`
to 
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { printf "%s\t",$3 }')`

Fix version checking for latest available version
Line 493 - Change
`padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')`
to 
`padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep -i 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')`

MatKier, 2020-05-26:
Ah yes, here is the original issue: https://github.com/jpmck/PADD/issues/76

KnownAim, 2020-05-30:
If you remove one "=" from each  of those lines, it fixes itself. Temporary fix for now!

plawer, 2020-06-19:
Any chance we will see those updates implemented soon?

derekschinke, 2020-07-03:
Updated to 3.2, and still have this exact same issue.

dschaper, 2020-07-12:
Please use #76

closed 2020-07-12 by dschaper