pi-hole/PADD #102: Has anyone tried PADD with Pi-hole v5?

christianboyle, 2020-05-19:
Wondering if any brave souls have attempted to use PADD with Pi-hole v5 yet.   
  
Any issues?   
  
Flawless install?  
  
Word is it's not easy to roll back to v4 due to the DB changes in v5.

downtime0, 2020-05-22:
I've had no issues with PADD v3.0.2 after updating to pi-Hole v5. Go for it!
However PADD v3.1 has a number of bugs in it so I recommend staying on v.3.0.x until the next minor v3.1.1 release.

downtime0, 2020-05-22:
@christianboyle hey I just implemented both your suggested code commits for v3.1 in my own file. Nice job on that. The fix for CleanEcho() was just what i was looking for. ...that and the fix for "padd_version_latest"

christianboyle, 2020-05-22:
[image]

[image]

Good stuff! Can confirm that stuff works well, haven't experienced any issues since I switched to using my [PR fork](https://github.com/jpmck/PADD/pull/85) (72d uptime, ellipsis works, etc.).   
  
Another issue thread has another set of fixes that includes [my fixes and others](https://github.com/jpmck/PADD/issues/86#issuecomment-598746643), which results in a more [full use of the display](https://github.com/jpmck/PADD/issues/86#issuecomment-597851806). Only issue seems to be the Memory bar not lining up with the CPU Temp.

downtime0, 2020-05-23:
Before last week's pi-Hole v5 update, mine was stable running like 100+ days as well.
...and today the clock starts over gain with PADD update.
[image]

Thanks for the link to the other thread, it didn't look like it was doing much more than I already had in my local file.

christianboyle, 2020-05-23:
That looks ace, any chance you could share your padd.sh?

closed 2020-05-23 by christianboyle

downtime0, 2020-05-23:
here you go.
[padd.sh.txt](https://github.com/jpmck/PADD/files/4670989/padd.sh.txt)

christianboyle, 2020-05-23:
thanks sir!