pi-hole/PADD #104: Pi-hole with PADD should not clutter up /var/log/auth.log

denott, 2020-05-24:
My pi-hole system's auth.log was about 38 MB per day, which seems excessive.

Upon reviewing the log file, it appears that PADD runs 'pihole status web' every few seconds, which causes the "pi" user to call 'sudo' each time.

> May 24 09:22:05 pi-hole sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/bash /usr/local/bin/pihole status web
> May 24 09:22:05 pi-hole sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> May 24 09:22:05 pi-hole sudo: pam_unix(sudo:session): session closed for user root
> May 24 09:22:10 pi-hole sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/bash /usr/local/bin/pihole status web
> May 24 09:22:10 pi-hole sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> May 24 09:22:10 pi-hole sudo: pam_unix(sudo:session): session closed for user root
> May 24 09:22:16 pi-hole sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/bash /usr/local/bin/pihole status web
> May 24 09:22:16 pi-hole sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> May 24 09:22:16 pi-hole sudo: pam_unix(sudo:session): session closed for user root
> May 24 09:22:21 pi-hole sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/bash /usr/local/bin/pihole status web
> May 24 09:22:21 pi-hole sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> May 24 09:22:21 pi-hole sudo: pam_unix(sudo:session): session closed for user root

The pihole "status web" command should not require root in order to run. (At least not on Raspbian Buster; it may very well be required when running on other distros and platforms. I think this might be better to address within Pi-hole, but would open a larger can of worms.)

One fix is to just run padd.sh as root. However from a security policy perspective, that may not be ideal.

The workaround I've done was to edit ~./padd.sh function GetPiholeInformation(): 
1. comment out the call to '$(pihole status web)' 
2. copy and paste from /usr/local/bin/pihole statusFunc()
3. snip and edit accordingly

> GetPiholeInformation() {
>   \# Get Pi-hole status
>   \#pihole_web_status=$(pihole status web)
> 
>   \# copied-and-pasted from /usr/local/bin/pihole:statusFunc()
>   \# Determine if service is running on port 53
>   if (echo > /dev/tcp/127.0.0.1/53) >/dev/null 2>&1; then
>     pihole_web_status=-1
>   fi
>   \# Determine if Pi-hole's blocking is enabled
>   if grep -q "BLOCKING_ENABLED=false" /etc/pihole/setupVars.conf; then
>     \# A config is commented out
>     pihole_web_status=0
>   elif grep -q "BLOCKING_ENABLED=true" /etc/pihole/setupVars.conf;  then
>     \# Configs are set
>     pihole_web_status=1
>   else
>    \# No configs were found
>    pihole_web_status=99
>  fi

With this change, my /var/log/auth.log file decreases to less than 40 KB per day. Much more manageable.

Related, but not as significant: during startup, PADD calls from GetVersionInformation(): "$(pihole -v -p)" and "$(pihole -v -a)" and "$(pihole -v -f)" which do unnecessary sudo calls. Replacing them with "$(/opt/pihole/version.sh -p)" and "$(/opt/pihole/version.sh -a)" and "$(/opt/pihole/version.sh -f)" reduces the sudo calls.

Perhaps the better place to address this would be in /usr/local/bin/pihole and have it not run sudo for options that don't require running as root. But I'm not sure if that would break things on other distros/platforms. Putting a workaround in PADD may break fewer systems, since users would have less variation between distros/platforms.

[image]

[image]

[image]

dschaper, 2020-07-12:
> The pihole "status web" command should not require root in order to run.

It does, any call to the `pihole` utility will require `sudo` or root privs. It's the controller for the DNS which requires privs to run on port 53.

subnut, 2021-08-13:
Use this -
```sh
sudo tee /etc/sudoers.d/020_pihole >/dev/null <<EOF
Cmnd_Alias PIHOLE = $(command -v pihole)
Defaults!PIHOLE syslog_goodpri=none, !pam_session
EOF
sudo chmod 0440 /etc/sudoers.d/020_pihole
```

Now there should not be any entries in `/var/log/auth.log`

thomasmerz, 2022-02-02:
@denott , @dschaper 
Is this fixed yet or why do I not see any of those messages in my Pi-Hole running in a Docker Container?
```
root@pihole-hel1-dc2:/# grep "pihole status web" /var/log/* -iR
root@pihole-hel1-dc2:/#
```

subnut, 2022-02-02:
@thomasmerz Maybe `pihole` is running as root?

What's the output of this -
```sh
ps -o uid,user,comm `pgrep hole`
```

thomasmerz, 2022-02-02:
That's right! I didn't took care of this in relation to this PR… 🤦🏻‍♂️ 

Output is (padd.sh is currently not running anymore):
```
🇩🇪  ✔ ~ [master|✔]
13:07 # pgrep -fla pihole -U root
4184 /usr/bin/docker start -a pihole
4539 /usr/bin/sh -c /usr/bin/docker exec -it pihole pihole -c > /dev/tty11 < /dev/tty11
4540 /usr/bin/sh -c /usr/bin/docker exec -it pihole pihole -t > /dev/tty10 < /dev/tty10
4542 /usr/bin/docker exec -it pihole pihole -t
4546 /usr/bin/docker exec -it pihole pihole -c
4788 bash /usr/local/bin/pihole -t
4846 bash /usr/local/bin/pihole -c
4860 tail -f /var/log/pihole.log
4896 bash /opt/pihole/chronometer.sh
5292 s6-supervise pihole-FTL
🇩🇪  ✔ ~ [master|✔] docker exec pihole ps -o uid,user,comm | pgrep hole
5324
```

subnut, 2022-02-02:
@thomasmerz I told you to run -
```sh
ps -o uid,user,comm `pgrep hole`
```
But you ran -
```sh
ps -o uid,user,comm | pgrep hole
```

thomasmerz, 2022-02-02:
My fault, I'm sorry that I misunderstood. Here we go:
```
root@pihole-nbg1-dc3:/# ps -o uid,user,comm `pgrep hole`
  UID USER     COMMAND
  999 pihole   pihole-FTL
```

thomasmerz, 2022-02-19:
@subnut , do have any update on my answers above and my question:
> Is this fixed yet or why do I not see any of those messages in my Pi-Hole running in a Docker Container?

subnut, 2022-02-19:
@thomasmerz Sorry, totally forgot :sweat_smile: 

My guesses are -
+ Maybe your `sudo.conf` does not log?
+ Maybe the user `pihole` is equivalent to root? (I don't know _anything_ about docker)

thomasmerz, 2022-02-19:
pihole has uid=999 and is not equal to root. And "User pihole is not allowed to run sudo …" in my docker container.  
❓ The main question is, if anybody still has this problem? The issue is quite old (2020) and many updates have been made since then. So maybe, problem is already fixed "silently"… 🤔  @subnut , @denott , please respond.

yubiuser, 2022-06-29:
This issue is fixed two-fold:
1) We do not require `sudo` anymore to get `pihole status (web)` and `pihole -v` since PR https://github.com/pi-hole/pi-hole/pull/4663
2) We do not run `pihole status web` anymore but make a direct call to `FTL` since https://github.com/pi-hole/PADD/pull/209

closed 2022-06-29 by yubiuser