pi-hole/PADD #105: PADD missing information and Screen Resolution

vekzla, 2020-05-31:
Hi PADD team,
I was forced to do a clean install of all my applications on RASPI which included PADD.
Has there been a new change to PADD to 3.1?
left screen shot is today and right is previous please ignore the out of screen proportion this has been fixed and captured as an error before.
The issue on the left - 
1) PADD title not coloured
2) I cannot get the text/content to fill the 7inch LCD
3) it seems to be missing information such as DHCP etc (please compare both screen grabs)
is this due to screen size thinking it is too small to display all the information?
let me know if you require more information
Thank you so much

[image]

KnownAim, 2020-06-02:
You could check out my commit where I made a few changes. I took code suggestions from comments in previous issues:
- Forced Mega size (line 860. can be deleted)
- Removed one "=" from lines: 636, 646, 649, 653, 673, 704, 715, 722, 748 and 759 to fix formatting issue (probably not needed for your screen. mine is 3.5")
- Fixed version check error (line 493)
- Fixed DHCP printing issue and artifact cleaning (line 331 and 570)

To try it, you would do the same installation method but with this URL
`cd ~
wget -N https://raw.githubusercontent.com/KnownAim/PADD/master/padd.sh`

Then to make it runable
`sudo chmod +x padd.sh`

vekzla, 2020-06-02:
> You could check out my commit where I made a few changes. I took code suggestions from comments in previous issues:
> 
> * Forced Mega size (line 860. can be deleted but you have a 7" screen and mega is needed)
> * Removed one "=" from lines: 636, 646, 649, 653, 673, 704, 715, 722, 748 and 759 to fix formatting issue (probably not needed for your screen. worth a try regardless. Otherwise can add one "=" back.)
> * Fixed version check error (line 493)
> * Fixed DHCP printing issue and artifact cleaning (line 331 and 570)
> 
> To try it, you would do the same installation method but with this URL
> `cd ~ wget -N https://raw.githubusercontent.com/KnownAim/PADD/master/padd.sh`
> 
> Then to make it runable
> `sudo chmod +x padd.sh`

thank you for your assistance. i have completed as per above however screen resolution still small and PADD is looking a bit weird with information missing as per the pic
[image]

vekzla, 2020-06-02:
ok figured it out, seems that i had to force my resolution in order for PADD to pick up what screen I have, hope that makes sense
framebuffer width 1920
framebuffer height 1080
worked for 7nch LCD

KnownAim, 2020-06-03:
Awesome! I'm glad to hear you figured out how to make it work. I'd love to see a final pic of the fixed version.

vekzla, 2020-06-03:
> Awesome! I'm glad to hear you figured out how to make it work. I'd love to see a final pic of the fixed version.

thank you for your assistance, see picture attached
[image]

closed 2020-06-03 by vekzla