pi-hole/PADD #108: Faulty Screendetection in 3.2

Gothico74, 2020-07-04:
After upgrade to PADD 3.2 the screen on my raspberry is teared up
way too much "="
TFT is 3.5 inch
Last padd 3.1 was working perfect with the two community fixes

[image]

LPanic, 2020-07-04:
yup. having the exact same issue since update.

LPanic, 2020-07-04:
Is there a way to downgrade to the previous version?

derekschinke, 2020-07-04:
Same issue.

Gothico74, 2020-07-04:
> 
> 
> Is there a way to downgrade to the previous version?

https://github.com/jpmck/PADD/releases

GorramRob, 2020-07-05:
I diffed 3.1 and 3.2, it looks like the issue may be line 860.

I commented it out and have not observed any problems so far, both on a similar pi screen to what is being used in the OP picture and on a normal terminal.

jpmck, 2020-07-05:
Fixed in v3.2.1.

closed 2020-07-05 by jpmck