pi-hole/PADD #110: Version check fails since repository move

snarkbe, 2020-07-10:
The version check fails again, although I'm running the latest version 

This is caused by the move of the PADD repository from https://github.com/jpmck/PADD/ to https://github.com/pi-hole/PADD/ as 2 redirections are now needed, instead of one before.

Changing [line 493](https://github.com/pi-hole/PADD/blob/09e34ae56892a61518cf8cba854c40501924caf8/padd.sh#L493) to:
` padd_version_latest=$(curl -sI https://github.com/pi-hole/PADD/releases/latest | awk -F / 'tolower($0) ~ /^location:/ {print $NF; exit}' | tr -d '\r\n[:alpha:]')`
solves the issue.

fireheadman, 2020-07-10:
thanks....  I provided some additional info (screenshots) if you need them.

PromoFaux, 2020-07-10:
OK, I guess we need to make the change and tag a new version to fix this.  @snarkbe would you like to make a pull request with the proposed change?

@jpmck what's the standard way for releasing on this repo? Is it PRs into `development`, then `master`? Or just straight into `master`?

peatrick, 2020-07-10:
Awesome, thanks you @snarkbe (et al).  This resolved things on both my Ubuntu Server 20.04 VM as well as my RPi3B+ running Raspbian GNU/Linux 10 Lite (buster).  Much obliged.

dschaper, 2020-07-10:
Or use the same version check that the Core uses? https://github.com/pi-hole/pi-hole/blob/4d25f695267590b61a4061f9bb43448005d99b85/advanced/Scripts/updatecheck.sh#L52

HinksUK, 2020-07-11:
I changed the URL to https://github.com/pi-hole/PADD/.. and it worked.

PR #111 (2020-07-12): Use GitHub API for remote version detection.
Use the GitHub API for remote version detection. 

Bumps to version v3.2.2.

Closes #73, closes #82, closes #90, closes #107 
Resolves #110, resolves #97, resolves #73, resolves #80, resolves #87
closed 2020-07-12 by dschaper