pi-hole/PADD #112: Certain stats not properly displaying in PADD v3.2.2

mokrunka, 2020-07-16:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
under the 'STATS' section, Blocking shows '0 domains', Clients is blank, Latest is 0, Top Ad is blank, Top Dmn is blank, Top Clnt is blank. I have reason to believe that pi-hole itself is working since the web UI is updating and telling me ads are actively being blocked. This seems to be an issue with how the data is collected for printing to the screen in PADD.

**To Reproduce**
Steps to reproduce the behavior:
1. run ./padd.sh
2. observe missing information

**Expected behavior**
blocked ads and other data should be properly reported, and not blank.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Barokai, 2020-07-16:
same for me, happens since the update to 5.1.1

snarkbe, 2020-07-16:
The FTL component doesn't write the file pihole-FTL.port anymore with the correct port to access the API. PADD uses this file to get the port to connect to the FTL API and retrieves the stats.

I changed [line 112](https://github.com/pi-hole/PADD/blob/master/padd.sh#L112) and hardcoded the port:
`ftl_port=4711`

This is a workaround of course. It should be solved the proper way, by dynamically retrieving the port. But each time I restarted FTL, it was always using port 4711 though.

Fleegle61, 2020-07-16:
Tried changing line 112.  Didnt see any difference it results when I ran PADD again.

snarkbe, 2020-07-16:
> Tried changing line 112. Didnt see any difference it results when I ran PADD again.
Changed to what? 
It works fine for me! 
[image]

Fleegle61, 2020-07-16:
Strange.  I changed padd.sh to have line 112 use port 4711 as noted above.  Only on my 1st cup of java this morning.  Let me go back and double check..

Fleegle61, 2020-07-16:
rebooted and still the same...Maybe I am not using port 4711?  Gotta dig a little deeper on this
[image]

snarkbe, 2020-07-16:
> Maybe I am not using port 4711? Gotta dig a little deeper on this

 `sudo lsof -Pi4 | grep pihole-FT`

and check the port listening on localhost only:
`pihole-FT 31642      pihole   13u  IPv4 4830919      0t0  TCP localhost:4711 (LISTEN)`

Fleegle61, 2020-07-16:
Fat finger!  I used /padd.sh instead of ./padd.sh (since I had accidentally downloaded padd.sh into root folder and not user folder)

[image]

mokrunka, 2020-07-16:
To make things a bit more robust, users can add an entry to /etc/pihole/pihole-FTL.conf as follows:

FTLPORT=4711

I think then the port number could be hardcoded into PADD.sh. I've spent some time looking through the source, but it's not clear to me exactly which code specifically assigns port 4711 for use.

piholeDebug.sh has a function in there to check the ports required for operation are available though - maybe something like this could be used to extract the port number for PADD? https://github.com/pi-hole/pi-hole/blob/56cd7c4d59ceb629a2170cd1c8ab78ef26366019/advanced/Scripts/piholeDebug.sh#L689

snarkbe, 2020-07-16:
> piholeDebug.sh has a function in there to check the ports required for operation are available though - maybe something like this could be used to extract the port number for PADD?

It still has a reference to the now-obsolete port file:
`FTL_PORT="${RUN_DIRECTORY}/pihole-FTL.port"`

mokrunka, 2020-07-16:
> > piholeDebug.sh has a function in there to check the ports required for operation are available though - maybe something like this could be used to extract the port number for PADD?
> 
> It still has a reference to the now-obsolete port file:
> `FTL_PORT="${RUN_DIRECTORY}/pihole-FTL.port"`

I was referring specifically to the notion of using lsof to source the port information for FTL.

dschaper, 2020-07-16:
Until I can get FTL fixed the below will work. Don't modify PADD, no changes are needed there...

`echo 4711 | sudo tee /var/run/pihole-FTL.port`

pheiduck, 2020-07-16:
After reboot its showing agian `FTL offline`

dschaper, 2020-07-16:
Yeah, `/var/run` is a temporary filesystem that doesn't survive reboot.

pheiduck, 2020-07-16:
hm so we have to wait for a Fixed Version?

dschaper, 2020-07-16:
Or run the command again.

mokrunka, 2020-07-16:
you can also edit (at your own risk) the ./padd.sh script on line 112 to read:
`ftl_port=4711`
until the pi-hole source is updated to once again write the port to the pihole-FTL.port file.

pheiduck, 2020-07-16:
is there a way to hide the echo of `4711` ?

snarkbe, 2020-07-16:
`echo 4711 >/var/run/pihole-FTL.port`

dschaper, 2020-07-16:
That only works if you're already root.

PureFallen, 2020-07-16:
> is there a way to hide the echo of `4711` ?

> `echo 4711 >/var/run/pihole-FTL.port`

won't work, even if you `sudo echo`.
If you are worried about echo being displayed on the PADD on display later: This should be gone as soon as PADD refreshes, especially since you only need to run this echo once each time the system is restarted.

One option to hotfix that for the moment could be to create that file in the reboot before ./padd.sh is executed.

dschaper, 2020-07-16:
I've already got the update to FTL submitted as a PR, I'm waiting for confirmation that it works and doesn't break something else. The rest of the team will be able to do that when they have a few moments or are awake in their respective time zones.

Fleegle61, 2020-07-17:
Gotta love the quick response!  Thanks Dan

vekzla, 2020-07-17:
[image]
Same here boys and girls, happened post pihole -up

dschaper, 2020-07-17:
Is this showing anything different than the previous 23 posts already covered?

vekzla, 2020-07-17:
> Is this showing anything different than the previous 23 posts already covered?

Not at all mate,

PureFallen, 2020-07-17:
> Not at all mate,

The "hotfix" solution is already digged under the existing replies, and that screenshot doesn't really help finding it any faster for the next one arriving here.

As said from @dschaper before:
> Until I can get FTL fixed the below will work. Don't modify PADD, no changes are needed there...
> 
> `echo 4711 | sudo tee /var/run/pihole-FTL.port`

I think that is everything that needs to be said.

dschaper, 2020-07-17:
Following up: The fix is set for the next release of FTL but there's a couple of more FTL issues we'd like to get in the same release. Since there's a quick workaround for this cosmetic issue (see the post directly above this one) we think it's okay to wait a day or two.

dschaper, 2020-08-09:
Sorry this took a lot longer than I intended but it should now be resolved.

dschaper, 2020-08-17:
Should all be working now.

closed 2020-08-17 by dschaper