pi-hole/PADD #115: PADD 3.2.2 w/ Pihole 5.1 doesnt show version

Jarsky, 2020-07-18:
**Describe the bug**
versions missing from 'piHoleVersion' file

**To Reproduce**
Latest version of Raspbian OS 
PADD version 3.2.2
Pihole 5.1.1

Run PADD, and shows no versions. 
Check piHoleVersion file, shows no versions.
core_version=
web_version=
ftl_version=
padd_version=v3.2.2

**Expected behavior**
Should add the current versions to the piHoleVersion file, to display on PADD output

**Screenshots**

[image]
[image]

**Additional context**

pi@elementsdns:~/.padd $ pihole -v
  Current Pi-hole version is v5.1.1.
  Current AdminLTE version is v5.1.
  Current FTL version is v5.1.

dschaper, 2020-07-18:
Have you made any changes to the script? PADD v3.2.2 shouldn't be working at all with FTL v5.1.

Jarsky, 2020-07-18:
> Have you made any changes to the script? PADD v3.2.2 shouldn't be working at all with FTL v5.1.

Nope haven't made any changes. My RPi automatically runs updates every Friday so it updated to Pi-hole v.5.1 automatically this week. I just grabbed the latest padd.sh file and replaced my existing one. 

GetVersionInformation() doesnt look to be related to FTL service.

`GetVersionInformation() {
  # Check if version status has been saved
  if [ -e "piHoleVersion" ]; then # the file exists...
    # the file exits, use it
    # shellcheck disable=SC1091
    source piHoleVersion

    # Today is...
    today=$(date +%Y%m%d)

    # was the last check today?
    if [ "${today}" != "${last_check}" ]; then # no, it wasn't today
      # Remove the Pi-hole version file...
      rm -f piHoleVersion
    fi

  else # the file doesn't exist, create it...
    # Gather core version information...
    read -r -a core_versions <<< "$(pihole -v -p)"
    core_version=$(echo "${core_versions[3]}" | tr -d '\r\n[:alpha:]')
    core_version_latest=${core_versions[5]//)}

    if [[ "${core_version_latest}" == "ERROR" ]]; then
      core_version_latest=${core_version}
      core_version_heatmap=${yellow_text}
    else
      core_version_latest=$(echo "${core_version_latest}" | tr -d '\r\n[:alpha:]')
      # is core up-to-date?
      if [[ "${core_version}" != "${core_version_latest}" ]]; then
        out_of_date_flag="true"
        core_version_heatmap=${red_text}
      else
        core_version_heatmap=${green_text}
      fi
    fi

etc.........

    # write it all to the file
    echo "last_check=${today}" > ./piHoleVersion
    {
      echo "core_version=$core_version"
      echo "core_version_heatmap=$core_version_heatmap"
`

Fede068, 2020-07-18:
I've exact the same issue.

dschaper, 2020-07-18:
Odd. So you have a value in `/var/run/pihole-FTL.port`?

Fede068, 2020-07-18:
Yes, 4711

chrisausbun, 2020-07-18:
Looks like Conditional Forwarding is also still showing as Disabled after enabling it and rebooting. I also ran `echo 4711 | sudo tee /var/run/pihole-FTL.port` after the reboot but still no change.

Josemilla, 2020-07-30:
I've exact the same issue.

Jarsky, 2020-08-17:
Working for me now with the latest v5.2 update

[image]

dschaper, 2020-08-17:
If we can get another confirmation or two then we'll close out this issue.

zakdeutsch, 2020-08-17:
Working ok here as well

chrisausbun, 2020-08-17:
Yep, I can confirm it’s working for me as well. Feel free to close out.

closed 2020-08-17 by dschaper