pi-hole/PADD #119: Full screen support for Hyperpixel 4 displays?

cmdshft, 2020-08-31:
**Is your feature request related to a problem? Please describe.**
When displaying PADD on a Hyperpixel 4 display, there is extra space around PADD data that is not utilized. See attached photo.

**Describe the solution you'd like**
Would like to have the entire Hyperpixel 4 display used to display information. Purely an aesthetic concern.

**Additional context**
[image]

chrisjeb, 2020-09-01:
How big is the display?

cmdshft, 2020-09-02:
It’s 4”, 800x480.

Kerdic, 2020-09-02:
Do you have any suggestions for what other information you would like to see that would fill up the screen?

Or would you like the existing information spread out more, to fill up the screen?

Kerdic, 2020-09-02:
Could you run these commands to determine the character width of the screen?

tput cols
tput lines

From a guess, it appears to be about 20 characters wider than MEGA, and 5 rows longer. So somewhere close to 100x31 in terms of characters (not pixels).

My first thoughts on that extra space would be to remove the top client and top ad section, and move the Pi-holed percentage and count over to the left.

Then use that extra space on the right to have a top 10 ads and top 10 client list, or something along those lines if possible. So 10 rows of ads on top of 10 rows of clients, with a header for each, and a space between, so that would take up 23 rows of the black space on the right...

Anyway, top x number lists is what comes to mind to fill that space.

cmdshft, 2020-09-02:
I am not home right now but as soon as I am I can run those utils for sure.

As for the other question, I guess primarily just making the information bigger to fit the screen better would be the primary idea.

Kerdic, 2020-09-03:
Right on, you might look into just increasing the text size of the console font. If you just want the current data to fill in the screen more, maybe a larger font would be the way to go?

https://www.raspberrypi-spy.co.uk/2014/04/how-to-change-the-command-line-font-size

cmdshft, 2020-09-06:
btw, the results for the commands are:

tput cols: 131
tput lines: 46

pappcam, 2020-09-29:
I also have this screen and I'd also like PADD to fill the screen. My google-foo hasn't helped. Someone must have solved this somewhere.

giuliomagnifico, 2020-10-02:
If you're interested I've done it and I'm working on it. Here's my custom padd.sh to fill to HyperPixel 4 and display more infos:

[image]

dschaper, 2020-10-02:
Looking forward to the Pull Request to contribute back that code to the main project.

cmdshft, 2020-10-02:
That's awesome! Any ability to choose the "PADD" logo as well? I'm partial to the rainbow puke version lmao

giuliomagnifico, 2020-10-02:
Hmm no sorry, I'm not able to do, I'm not a developer, but it's easy to do if you want I can change it for you (just remove and add some comments lines # )

And if you want I can send you the code to test it, because I'm testing it on some different screens, all 800x480px and it works fine but, since it grabs hardware infos, I would test also on different devices than RPi 4  

If you want write me an email at giuliomagnifico@gmail.com and I'll send you the code to test it as an "alpha release".

giuliomagnifico, 2020-10-03:
Last interation of the script, now it shows also the CPU current speed VS the max speed. I don't know if its better to have:

`600 MHz / 1.50 GHz` or `4x @ 600 MHz`

It's an aesthetic question only I think :-) If anyone is interested in beta testing just write me at giuliomagnifico@gmail.com and I'll send the modified padd.sh file. Thanks!

[image]

dschaper, 2020-10-03:
You can set up your own repository then and provide the support and files with your fork of the project.

closed 2020-10-03 by dschaper