pi-hole/PADD #123: Pico being forced on mini pitft

mycomachine, 2020-09-09:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
Pico being displayed on 1.4", micro is proper version. ran sudo dpkg-reconfigure console-setup. selected 24x12 (framebufferonly) and it displayed correctly, but after a few reboots its back to displaying the pico version and if I run dpkg-reconfigure again I get the "PADD isn't for ants" message.

**To Reproduce**
Steps to reproduce the behavior:
1. install pitft option #6 270 degree rotation and latest PADD 3.3.1
2. observe version displayed

was also doing this on the 2.8" capacitive touch screen and pi2 b, was displaying mini, but tiny was correct (or vice versa)

[image]
[image]

fullburnen, 2020-09-13:
Looks like something is changing your font and size.

What's the contents of /etc/default/console-setup?

Jimbob-B, 2021-01-21:
Just found this project and had a same issue. After various attempts of tinkering with the console settings I ran `sudo dpkg-reconfigure console-setup` and left the font size the same as the boot console font - I now have the micro display in the first screenshot on my Pi Zero.

My `/etc/default/console-setup` looks like this,

`# CONFIGURATION FILE FOR SETUPCON`
`# Consult the console-setup(5) manual page.`
`ACTIVE_CONSOLES="/dev/tty[1-6]"`
`CHARMAP="UTF-8"`
`CODESET="guess"`
`FONTFACE=""`
`FONTSIZE=""`
`VIDEOMODE=`
`SCREEN_WIDTH=40`
`SCREEN_HEIGHT=18`
`# The following is an example how to use a braille font`
`# FONT='lat9w-08.psf.gz brl-8x8.psf' `

HTH

mycomachine, 2021-01-21:
Reboot it a few times, you'll see it happen again

Sent from ProtonMail mobile

-------- Original Message --------

closed 2022-12-17 by github-actions[bot]