pi-hole/PADD #130: Switch logo from PADD to Pi-Hole (like old) Cronometer

giuliomagnifico, 2020-09-21:
Hello,
I would ask if it's possible to insert a simply configuration to change the current logo to [the old one](http://i.imgur.com/gmkxsSn.jpeg), the same as it displays running Pi-Hole -c. Or can you point me where to edit the padd.sh file, because I'm able to delete and modify the current logo but I'm not finding the variable to print the old one.

Thanks a lot

[image]

dschaper, 2020-09-21:
They are two separate entities. Eventually `pihole -c` will go away and be replaced by PADD. 

For now there won't be any official changes, PADD will display PADD.

giuliomagnifico, 2020-09-21:
Okay thanks, I don’t want to use pihole -c but just the logo on PADD, can you point me on what to edit in padd.sh to replace the current logo with the Pi-Hole logo that appears running pihole -c? 

Thanks

dschaper, 2020-09-21:
Sorry, I can not.

closed 2020-09-25 by giuliomagnifico