pi-hole/PADD #134: PADD error after running for several hours

scuk23, 2020-10-02:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*
Newest version - non beta

**Describe the bug**
Everything starts and works perfectly. But, I'll come back and look at the display a few hours later and see this error.

**To Reproduce**
Steps to reproduce the behavior:
1. ... Start and run PADD.
2. ... Wait.

**Expected behavior**
Errors occur - pictured in screenshot.

**Screenshots**
https://i.imgur.com/caTrsnC.jpg

**Additional context**
I was asked to post this here. This discussion started on this reddit thread, https://www.reddit.com/r/pihole/comments/j3sjuj/padd_error_after_running_for_a_few_hours/

zangaby, 2020-12-01:
Same issue for me after I updated PADD to the latest version.

bitvac, 2020-12-04:
Same error generated for me, its displayed on screen 6 times.
Happend after 8 hours 30 mins up time.
Running: 
Pi-hole v5.2.1
Web Interface v5.2.1 
FTL v5.3.2 
PADD v3.4

kevin25761, 2020-12-10:
I am also receiving this error after installing a new 3.5" display  
https://www.amazon.com/gp/product/B07S695VQM/
http://www.lcdwiki.com/MHS-3.5inch_RPi_Display

Pi-hole 5.2.1
Web interface 5.2.1
FTL  5.3.2
PADD 3.4

Previously I was using a 2.8" Adafruit display and never received the error.
https://www.adafruit.com/product/3974

dasher7320, 2021-02-27:
Hi there,

After using PH with PADD for over 2 years, this issue has suddenly popped up, randomly every few days.

I can also confirm the CPU Temp increase, maxed out memory, screen looks the same as the top screenshot. I can connect with Putty and do the reboot command, restarts fine till next time it occurs, only warning i get is internet goes down, also note tonight's case happened at about the same time i disabled PiHole via the web interface for 5 mins to test a connection issue (to see if i needed to whitelist a domain).

Any updates on the fix?

Thanks

Dash

thomasmerz, 2022-02-28:
There's no problem at my installation running `padd.sh` for a week or sometimes two weeks (if no maintenance with restart/reboot was neccessary of Pi-hole or operation system)… 🤷🏻‍♂️ 

So is this still an issue for you?

gooperlooper, 2022-03-04:
Not me, but I recently had the same issue start up with me. No idea what's caused it.

closed 2022-12-17 by github-actions[bot]