pi-hole/PADD #136: Remove DHCP from output when DHCP Disabled

tmburton, 2020-10-06:
Enhancement Request:
If the DHCP server is disabled, remove DHCP from the PADD output/view.

Example:
DHCP ==========================================================================
 DHCP:     Disabled            IPv6 Spt: N/A
 Router:   10.10.10.1

thomasmerz, 2022-02-02:
Is this already fixed? Because I can't see anything regarding DHCP (because it's disabled? 😆 )

```
 __      __  __
|__) /\ |  \|  \  Pi-hole® v5.8.1, Web v5.10.1, FTL v5.13
|   /--\|__/|__/  PADD v3.6.3[✓] System is healthy.

PI-HOLE ===================================================
 Status:   Active              FTL:      Running
STATS =====================================================
 Blocking: 4185344 domains
 Pi-holed: [■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■■] 15.7%
 Pi-holed: 1161 out of 7399 queries
 Latest:   gsp-ssl.ls.apple.com
 Top Ad:   app-measurement.com
 Top Dmn:  spclient.wg.spotify.com
 Top Clnt: docker-br-4b9270b48dc1_pihole
NETWORK ===================================================
 Hostname: pihole-hel1-dc2     IP:       172.21.0.2
 IPv6:
 DNS:      4 servers           DNSSEC:   Enabled
SYSTEM ====================================================
 Uptime:   1 days, 5 hours, 58 minutes
 CPU Temp: 0.0°C               CPU Load: 0.62, 0.46, 0.31
 Memory:   [■■■■■·····] 54.0%  CPU Load: [··········] 0.0%
```

thomasmerz, 2022-02-19:
I like to know if DHCP is disabled. I personally dislike removing this info.

closed 2022-12-17 by github-actions[bot]