pi-hole/PADD #137: If Pi-Hole is running on VM don't show CPU Temp

tmburton, 2020-10-06:
When Pi-Hole is installed on a VM the CPU Temp is shown as "CPU Temp: 32.0°F"

Instead maybe we should show 0F/Unknown/VM or some other status.

Another option would be to remove CPU temp all together but unsure of what to put in that space or just have empty space.

tgjohnst, 2020-10-06:
I have created a PR to address this issue - please check it out and see if this is a fix that would address your feature request

tmburton, 2020-10-06:
Yes, I think this would work. Thank you

thomasmerz, 2022-02-19:
Why has the PR been closed without merging? I still see "CPU Temp: 0.0°C" on my VMs…

PromoFaux, 2022-02-19:
From the PR 

> I'm going to close this PR and make a cleaner one in its place once I solve it.

closed 2022-12-17 by github-actions[bot]

thomasmerz, 2023-05-30:
@tgjohnst , didn't you want to make a "cleaner" PR for this one? 🤔 I really would like not seing "CPU Temp: 0.0°C" anymore on my VMs.