pi-hole/PADD #138: Showing CPU Temp in C in PADD when Pi-Hole Web Interface is set to F

tmburton, 2020-10-06:
Running Pi-hole v5.1.2 Web Interface v5.1.1 FTL v5.2 and PADD v3.4 on a Raspberry Pi 4

Web interface is set to Fahrenheit

[image]

[image]

But PADD is showing Celsius:
CPU Temp: 44.8°C 

I believe that it should be showing Fahrenheit, is there something I am missing?

Thank you

tmburton, 2020-10-06:
I just noticed "Styling (auto saved, per-browser)" so that would appear its a client cookie that is saving it to be F and not in the actual Pi-hole config.

How can I resolve this?

dschaper, 2020-10-06:
It should also be in `/etc/pihole/setupVars.conf` as a variable.

tmburton, 2020-10-06:
Thank you, I added TEMPERATUREUNIT=F to /etc/pihole/setupVars.conf

I know its not a part of PADD per se but since PADD relies on the data, do you think it would be beneficial to add this information to the Wiki?

To change Pi-Hole temp reporting and for PADD to display the unit type desired modify the following:
/etc/pihole/setupVars.conf

For Fahrenheit add:
TEMPERATUREUNIT=F

For Celsius add:
TEMPERATUREUNIT=C

For Kevlin add:
TEMPERATUREUNIT=K

Restart the Pi-Hole service and then PADD for it to take effect
sudo systemctl restart pihole-FTL.service

closed 2020-10-07 by tmburton