pi-hole/PADD #140: Display OverlayFS "RW/RO" state

gsemet, 2020-10-11:
It would be really helpful to be able to display the current OverlayFS state (ReadOnly or ReadWrite) in the dashboard somewhere. Thanks

dschaper, 2020-10-11:
Try to figure out where to display it on the various sizes of screens and we can take a look. Screen mockups would be helpful, no need to write any code for it.

gsemet, 2020-10-11:
isn't it possible to have some place where user can display a few character easily? Or the only way to do so it to fork the project ?

dschaper, 2020-10-11:
If you want it done any time in the near future then yes, you'll need to do the work. 

I don't see a need to put that information on the display so I'm not really motivated to find a place for it. If you would like it then you'll need to show me where you think it looks the best on the display. And how it should work on the different sizes of displays.

closed 2021-01-25 by dschaper