pi-hole/PADD #141: Can't get screen to autorun on dietpi

alexlab963, 2020-10-24:
Hello,

I've finally decided to update chronometer to padd on my pihole setup.
I am having trouble finding how to setup padd so it runs on the attached screen after boot.
I have followed the directions and edited the .bashrc file but it doesn't seem to work, I also added the padd.sh script in the postboot folder, autorun file and custom.sh

What should I try next?

dschaper, 2020-10-25:
@michaing I don't see anything that would cause this to happen on DietPi, but if you could take a look I would appreciate it.

MichaIng, 2020-10-25:
Which hardware/SBC model is it?

Generally he setup instructions should work generic, starting the script once you login with the user on whichever screen you enter login cresentials, even on SSH this should work fine: https://github.com/pi-hole/PADD#setup-padd

`/var/lib/dietpi/postboot.d` is not made for long running processes but for one-time tasks, similar or a replacement for `/etc/rc.local`: https://github.com/MichaIng/DietPi/blob/master/rootfs/var/lib/dietpi/postboot.d/readme.txt

`custom.sh` autostart option is more for background processes. It prints to the TTY as well, but that is more aimed for script start info, not so much to have something shown and updated persistently on the screen: The TTY it prints to is not clearly defined (`StandardOutput=tty` systemd unit option) and since the STDIN (standard input) is not attached, you can go on using the console, which means that your console input and service output interfere.

So `~/.bashrc` is most reasonable here, so that it is executed once opening an interactive shell session with the current user (i.e. login) using STDIN and STDOUT of the current console/TTY you logged in, so you can terminate it easily by hitting ctrl+c. if required.
On DietPi you could use `/etc/bashrc.d/` to have it started on every user login, but does not make much sense here 😄.
`~/.profile` and `/etc/profile.d/` are as well possible, which are loaded on every shell (not just bash) but only on login, not when you e.g. open a subshell inside a shell, e.g. a shell emulator from desktop.

If you want to have it started without manual login (e.g. no keyboard attached), you can use `dietpi-autostart` > `7  : Automatic login`.

alexlab963, 2020-10-25:
My Dietpi is set to auto-login already, I've put the starting script everywhere I can think of, it even starts when I connect to it via ssh.
However, it doesn't start on the connected screen.

MichaIng, 2020-10-25:
> However, it doesn't start on the connected screen.

What do you mean with connected screen, is this the screen where the auto-login happens?

alexlab963, 2020-10-25:
> 
> 
> > However, it doesn't start on the connected screen.
> 
> What do you mean with "connected screen", is this the screen where the auto-login happens or is there another console?
> 
> If so, what happens if you manually load the script:
> 
> ```
> source ~/.bashrc
> ```
> 
> If nothing, please paste the content:
> 
> ```
> cat ~/.bashrc
> ```

I can run the ./bashrc script just fine and padd comes up, in fact, padd comes up every time I log in via ssh.
The problem is that it doesn't come up on the [waveshare](https://www.waveshare.com/img/devkit/LCD/3.2inch-RPi-LCD-B/3.2inch-RPi-LCD-B-7.jpg) screen that is attached to odroid c2. I can clearly see a terminal window that is already loged in

MichaIng, 2020-10-25:
> EDIT: I realized while writing this that the user with which it auto-logins might be the problem, I changed it to root and I can now see padd on my screen.

Ah yes, when using `~/.bashrc` is is for the user that you are currently logged in. Otherwise you can enable it for any user by either by logging in with that one to create the script/append the script or use `/home/<username>/.bashrc` directly. `~` us just a shortcut to the home dir of the current user.

> How can I make it full screen?

Is the console itself not full screen or PADD only? Not sure but it might simply have a fixed size so that you would need to reduce the resolution.

alexlab963, 2020-10-25:
Lowered the resolution to 480p and it works great now!
Thanks!

closed 2020-10-25 by alexlab963

MichaIng, 2020-10-25:
Great 🙂. Just checked the code and it adjusts itself to the screen size but only until a max width of 80 characters and height of 26 lines: https://github.com/pi-hole/PADD/blob/master/padd.sh#L896-L930

With default character dimensions 16x8 indeed 480p should work well:
8*80=640
16*26=416