pi-hole/PADD #145: Router IP has an "ex" after the ip

giuliomagnifico, 2020-12-25:
Hi, since the 5.0 PADD displays an `ex` after my router ip.

Running Pi-Hole on a Raspberry PI 4B (infos in the photo)

[image]

It appears not right after the start but after some time (sometimes 10 minutes, sometimes 10 days). Looks like it grabs the correct IP but for some reasons it add/fill 2 spaces with random characters after the IP address, usually when the router is rebooted and Pi-Hole is keep running. This is not happening if I runs PADD in my terminal via ssh. Looks like a weird bug and is not related to my custom PADD setup, it was appearing also with the default script, and I haven't touched the part of code when it grabs the router ip.

I solved by manually entry my router ip but is only a workaround. I tried to add two empy spaces after the 

 ` dhcp_info=" Router:  ${DHCP_ROUTER}  "` 

but doesn't work, the "ex" appears again.

```
# if the DHCP Router variable isn't set
    # Issue 3: https://github.com/jpmck/PADD/issues/3
    if [ -z ${DHCP_ROUTER+x} ]; then
      DHCP_ROUTER=$(/sbin/ip route | awk '/default/ {printf "%s\t", $3}')
    fi
    dhcp_info=" Router:   192.168.1.2" #${DHCP_ROUTER}"
    dhcp_heatmap=${red_text}
    dhcp_check_box=${check_box_bad}
```

yubiuser, 2022-06-28:
Not sure if this is still an issue. 
But with https://github.com/pi-hole/PADD/pull/235 we change the way to get the information, which might or might not solve this issue (if it ever was connected to how we got `DHCP_ROUTER`

yubiuser, 2022-12-04:
Closing due to inactivity. Feel free to reopen if issue persists.

closed 2022-12-04 by yubiuser