pi-hole/PADD #147: pihole-ftl v.5.3.4 / PADD v3.4 - telnet_connection_handler_thread

un99known99, 2020-12-28:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
pihole-ftl.log is filling up with massive warnings

**To Reproduce**
Steps to reproduce the behavior:
1. start padd.sh (version 3.4)
2. tail -f on pihole-ftl.log

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[2020-12-28 13:48:56.967 19626/T24654] WARN: Could not recv() in telnet_connection_handler_thread() (/root/project/src/api/socket.c:319): Connection reset by peer
[2020-12-28 13:48:56.987 19626/T24657] WARN: Could not recv() in telnet_connection_handler_thread() (/root/project/src/api/socket.c:319): Connection reset by peer

un99known99, 2020-12-28:
using official versions:
Pi-hole® v5.2.2, Web v5.2.2, FTL v5.3.4, PADD v3.4

DL6ER, 2020-12-29:
PADD uses the same method for connecting to FTL as `chronometer` does:

https://github.com/pi-hole/pi-hole/blob/41bdb741b7e1a70184ecd7dbfc23f3ebcaa5e3bc/advanced/Scripts/chronometer.sh#L15-L40

https://github.com/pi-hole/PADD/blob/1c93c02a36140d30cdc14369e27a592b70cf47f5/padd.sh#L119-L148

(both on current branch `master`)

As you can see, what is reported by FTL is correct: The connection is just closed at the end without sending FTL the `>quit` command. Hence, FTL expects something more to come through the pipe and is then surprised by a suddenly closed connection.

This can be solved in two ways:
1. Terminate the connection properly (i.e., send `>quit` to FTL) when the connection is not needed any longer
2. Ignore the client's connection reset (as we did before FTL v5.3.3)

1 seems to be the preferred variant and is proposed in https://github.com/pi-hole/pi-hole/pull/3954

closed 2020-12-30 by dschaper

un99known99, 2020-12-31:
not fixed yet , pihole -c still showing errors

DL6ER, 2020-12-31:
This issue is about `padd` which was already updated and fixed. `pihole -c` is fixed as well, however, the fix will only be distributed with the next release of Pi-hole.