pi-hole/PADD #148: PADD Docker?

AndyVRD, 2020-12-28:
Hi,

Is there any solution using PADD if Pihole is running in a container? I keep getting a error...

cat: /var/run/pihole-FTL.port: No such file or directory

DL6ER, 2020-12-29:
You can create the file and put `4711` into it as the port is fix.

@dschaper Now that PADD is under the Pi-hole namespace, maybe we can just include it in the official container?

AndyVRD, 2020-12-29:
> You can create the file and put `4711` into it as the port is fix.
> 
> @dschaper Now that PADD is under the Pi-hole namespace, maybe we can just include it in the official container?

Thanks for the reply. Just changed the   # Get FTL port number ftl_port=$(cat /var/run/pihole-FTL.port) to   # Get FTL port number ftl_port=$(cat /var/run/pihole-FTL.4711) but still same error.

DL6ER, 2020-12-29:
Try maybe something like

```
exec 3<>"/dev/tcp/localhost/4711"
```

you accidentally removed a `$` in front of the variable name

PromoFaux, 2020-12-29:
Also if I'm not mistaken you'll need to expose port 4711 on the docker container. What does your compose file / docker run look like?

AndyVRD, 2020-12-29:
Sorry i'm not that good with this so any change to explain where i can find this compose file? Changed the code like DL6ER say and now i got indeed a connection refused error.

./padd.sh: connect: Connection refused
./padd.sh: line 118: /dev/tcp/localhost/4711: Connection refused

PromoFaux, 2020-12-29:
What I mean is, how did you start the docker container? I'm guessing, from context, that you ran a command like `docker run...`?

AndyVRD, 2020-12-29:
> What I mean is, how did you start the docker container? I'm guessing, from context, that you ran a command like `docker run...`?

I used [this guide](https://homenetworkguy.com/how-to/install-pihole-on-raspberry-pi-with-docker-and-portainer/). :)

dschaper, 2020-12-29:
It won't run in a separate docker container. You have to have access to the socket in the `pihole` container and exposing that is not a good idea since there is no verification/validation for access.

dschaper, 2020-12-29:
> Now that PADD is under the Pi-hole namespace, maybe we can just include it in the official container?

Sure, we could. I don't know how useful it would be since it needs a terminal and most of the time docker is not started with an interactive terminal. I guess you could `docker exec -it` and then run PADD but that's a lot of tinkering.

closed 2021-01-25 by dschaper