pi-hole/PADD #157: Pi-Hole version is not detected and displayed

slodown, 2021-03-04:
Hi,

I'm using Pi-hole v5.2.4 / Web Interface v5.4  / FTL v5.7 and PADD 3.5.1. 

During verifying at startup the version could not be detected.

[image]

Many thanks!

dschaper, 2021-03-04:
What does `pihole -v` return for versions?

slodown, 2021-03-04:
```
pi:~ $ pihole -v
  Current Pi-hole version is v5.2.4.
  Current AdminLTE version is v5.4.
  Current FTL version is v5.7.
```

dschaper, 2021-03-05:
Stop PADD, remove `piHoleVersion` and then restart and see if that clears out.

slodown, 2021-03-05:
This has not changed the behavior.

slodown, 2021-03-05:
I do not know why, but now it is showing the version information.

closed 2021-03-05 by slodown