pi-hole/PADD #158: PADD not displaying information

LuxStatera, 2021-04-01:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
A clear and concise description of what the bug is.

I dont think this is a bug but I am not sure why my display looks different than all the pictures people post. My display doesn't show the long green and red bar, the temp, the groups like system and network. it shows all the standard info but not everything. so im not sure if this is just how the new version looks or if i did something incorrectly.
**To Reproduce**
Steps to reproduce the behavior:
1. installed PADD on my raspberry pi 0 W that already had pihole on it
2. setup hdmi for PADD
3. reboot system to start PADD
4. It displays the information and is running and active but doesnt show ALL the information

**Expected behavior**
A clear and concise description of what you expected to happen.
i expected my display to look the same as the picture on github
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

dschaper, 2021-04-02:
What are the dimensions of the screen? It looks like you have the Tiny display.

LuxStatera, 2021-04-02:
"UCTRONICS 7 Inch IPS Touchscreen for Raspberry Pi 1024×600 Capacitive HDMI LCD Monitor Portable Display"

is there a way to switch to the normal display?

LuxStatera, 2021-04-02:
[image]

dschaper, 2021-04-02:
Size display is determined with a couple of `tput` calls:

https://github.com/pi-hole/PADD/blob/master/padd.sh#L1201

What is the output from:

```bash
tput cols
tput lines
```

LuxStatera, 2021-04-02:
cols 81
lines 38

dschaper, 2021-04-02:
Sorry, that display isn't PADD. Where did you download the script from?

LuxStatera, 2021-04-02:
i followed the "Setup PADD" on the README.md

dschaper, 2021-04-02:
Then something is not configured correctly. The default display shows "PADD" not "Pi-hole". 

Have you made any modifications?

LuxStatera, 2021-04-02:
i think i made an adjustment for hdmi. ill try to start fresh. how would i go about uninstalling it?

dschaper, 2021-04-02:
There's nothing to install or uninstall. It's just a script that is run via `bash`.

LuxStatera, 2021-04-02:
sorry im a noob. i appreciate the help. im going to reflash the sd card and start fresh

dschaper, 2021-04-02:
You don't need to do all that. It's just a script call `padd.sh` in your home directory.

LuxStatera, 2021-04-02:
i deleted it and removed the setting from the ./bashrc but it still pops up after a reboot

dschaper, 2021-04-02:
Did you follow another persons guide or use another persons code like (https://gist.github.com/giuliomagnifico/39f9912a078418c1fdcb36d588ab3630?

LuxStatera, 2021-04-02:
[image]

LuxStatera, 2021-04-02:
followed this page. i am using a raspberry pi zero w if that changes things

dschaper, 2021-04-02:
The problem is that the logo you are showing just doesn't exist in this repo's code. It does exist in that code that I linked.

LuxStatera, 2021-04-02:
[image]

LuxStatera, 2021-04-02:
ive seen it in a few other places so i assumed it was correct. not sure what i did

dschaper, 2021-04-02:
That's not PADD.

Do you have any files that end in .sh in your home directory?

LuxStatera, 2021-04-02:
nothing in my home directory

LuxStatera, 2021-04-02:
i might be mistaken but if its the main directory i boot to then its also empty

dschaper, 2021-04-02:
Then I'm not understanding how a file that doesn't exist can be started...

LuxStatera, 2021-04-02:
its all good. im just gonna reflash it and try again. once again thank you for the help

LuxStatera, 2021-04-02:
started fresh. reflashed. installed and setup pihole. "cd ~" then "wget -N https://raw.githubusercontent.com/pi-hole/PADD/master/padd.sh" then made it executable and setup auto run. i reboot my pi and it just brings me to the command line and PADD doesnt show. am i doing something wrong?

LuxStatera, 2021-04-02:
[image]

closed 2021-04-02 by LuxStatera

dschaper, 2021-04-02:
Great!

jwebbstevens, 2022-11-04:
I know this issue is closed but I have the same setup.

It's important to make sure the Raspberry Pi 4 (etc) boots to the Command Line.

`raspi-config` and then follow the prompts as follows:

1. Select Menu Option 1 called _System Options_
2. Select Menu Option S5 called _Boot / Auto Login_
3. Select Menu Option B2 called _Console Autologin_

what this does is on boot the pi will always login automatically to the Console.

This assumes everything else is setup correctly including HDMI0 as the main interface to the screen.