pi-hole/PADD #159: Add Unbound Stats

bakerboy448, 2021-04-23:
**Is your feature request related to a problem? Please describe.**
Add unbound stats
**Describe the solution you'd like**
Implement 
https://gist.github.com/giuliomagnifico/e5b52263547c130bdfea483929c0689e

**Describe alternatives you've considered**
None
**Additional context**
https://giuliomagnifico.medium.com/add-unbound-stats-to-padd-for-pi-hole-38b2ea87db2b

thomasmerz, 2022-02-28:
That's really nice 😃 But only valid for `unbound` users and not for those with `dnsmasqd` 😞 

You really should make a pull request and put [these lines](31007) in a condition that checkes if `unbound` is running (or `unbound-control` is a valid command. For example: `if type -p unbound-control; then … fi`).

Does anyone know if there are also some nice stats for `dnsmasqd`? I couldn't find anything nice other than `sudo pkill -USR1 dnsmasq` which doesn't work on my (dockerized) Pi-hole… 🤔

thomasmerz, 2022-03-04:
I found accidentally this in `/var/log/pihole-FTL.log` - this could be used for this, couldn't it? (only "reply" (in seconds) is missing)
```
[2022-03-05 00:08:56.894 1084M] Imported 39729 queries from the long-term database
[2022-03-05 00:08:56.895 1084M]  -> Total DNS queries: 39729
[2022-03-05 00:08:56.895 1084M]  -> Cached DNS queries: 21259
[2022-03-05 00:08:56.895 1084M]  -> Forwarded DNS queries: 11789
[2022-03-05 00:08:56.895 1084M]  -> Blocked DNS queries: 6340
[2022-03-05 00:08:56.895 1084M]  -> Unknown DNS queries: 299
[2022-03-05 00:08:56.895 1084M]  -> Unique domains: 2779
[2022-03-05 00:08:56.895 1084M]  -> Unique clients: 17
[2022-03-05 00:08:56.895 1084M]  -> Known forward destinations: 2
```

yubiuser, 2022-06-28:
> Does anyone know if there are also some nice stats for dnsmasqd?

https://docs.pi-hole.net/ftldns/telnet-api/

thomasmerz, 2022-06-30:
Does this help? https://github.com/barreljan/dnsmasqstats

thomasmerz, 2022-06-30:
Or this thread? https://discourse.pi-hole.net/t/cache-server-statistics/6092

closed 2022-12-17 by github-actions[bot]