pi-hole/PADD #160: Wireguard status

thiefrainb0w, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
No Problem, feature request

**Describe the solution you'd like**
Padd is great and all, since more and more users seem to run a pihole/wireguard combo it would be nice to see the status of a Wireguard Interface somewhere on Padd too, maybe up top next to "Pi holes DNS" status. in a similar way. 

red dot means "Wg interface off"
yellow "on but no handshake"
green "handshake ok"

Wireguard also shows Tx/Rx traffic if that can be maybe added into the bigger Padd layouts.

**Describe alternatives you've considered**
Tried looking into customizing it myself, research will take a while

**Additional context**
Sometimes Wireguard cuts out on me, not giving me access into my home network. This feature could save time by seeing if the connection to another Wireguard instance is ok. Granted, it would not mean that Wireguard is set up correctly but still help rule out other problems (ex. is Wg interface even enabled?)

thomasmerz, 2022-02-19:
I'm also using a Pi-hole/Wireguard combo on my cloudserver(s), but both are running in different docker containers and different docker networks and both don't know about each other:
```
🇫🇮  ✔ ~ [master|✔]
18:27 # docker network ls
NETWORK ID     NAME                                      DRIVER    SCOPE
728880a1f19b   bridge                                    bridge    local
4b9270b48dc1   docker-pi-hole_default                    bridge    local
6dcb15ea146f   docker-wireguard-ubuntu-cx11-03_default   bridge    local
```

And in general I think that Wireguard is not related to PADD… What does Pi-hole team think about this?

yubiuser, 2022-02-19:
I think this is outside the scope of Pi-hole/PADD.

thomasmerz, 2022-02-19:
If we find more people that find this out-of-scope, then this issue can and will be closed? 🤔

thomasmerz, 2022-03-19:
@yubiuser , could you please close this out-of-scope issue? Thanks 👍🏻

closed 2022-03-19 by yubiuser