pi-hole/PADD #162: IPv4 not showing

grug0r, 2021-07-10:
**Describe the bug**
Under Network IPv4 address not showing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run ./padd.sh

**Expected behavior**
Display local Pi-hole server IPv4 address.

**Screenshots**
[image]

**Additional context**
Ubuntu 18.04.05 LTS running Pi-hole 5.3.1

DL6ER, 2021-07-10:
What is the output of
```
ip addr | grep 'inet '
```
?

You can randomize the digits for privacy reasons but please leave the overall structure intact so we can check what may be the issue.

grug0r, 2021-07-10:
Wow that was quick :)

root@pihole:~# ip addr | grep 'inet '
    inet 127.0.0.1/8 scope host lo
    inet 192.168.60.222/24 brd 192.168.60.255 scope global eth0
    inet 169.254.182.150/16 brd 169.254.255.255 scope global eth0

DL6ER, 2021-07-10:
Okay, that looks good, how about 
```
ip addr | grep 'inet ' | grep -v '127.0.0.1/8' | awk '{print $2}' | cut -f1 -d'/'
```

grug0r, 2021-07-10:
root@pihole:~# ip addr | grep 'inet ' | grep -v '127.0.0.1/8' | awk '{print $2}' | cut -f1 -d'/'
192.168.60.222
169.254.182.150

grug0r, 2021-07-10:
Of further info, padd IPv4 was displaying an inet6 address. To test I disabled IPv6 and hence where I'm at now.

grug0r, 2021-07-10:
I can enable IPv6 again and provide a screenshot if needed.

DL6ER, 2021-07-10:
Yeah, more information is always good. This is the problematic line of code (just for reference):

https://github.com/pi-hole/PADD/blob/1dcf62b2d50e55b8ed78b6e1dc52d2c6af77a7ea/padd.sh#L718

DL6ER, 2021-07-10:
Ah, I found a small bug, bugfix incoming...

grug0r, 2021-07-10:
Any further info you need or I can do on my end to quick fix?

grug0r, 2021-07-10:
Oh cool, thank you :)

DL6ER, 2021-07-10:
Could you try `padd.sh` from https://github.com/pi-hole/PADD/blob/fix/typov4v6/padd.sh ?

grug0r, 2021-07-10:
nope, still blank.

DL6ER, 2021-07-10:
Does enabling IPv6 still bring back the IP address in the wrong place?

grug0r, 2021-07-10:
Let me test, one moment.

grug0r, 2021-07-10:
After enabling IPv6..
 IPv4: fe80::b051:45ff:fec7:b165 IPv6:

DL6ER, 2021-07-10:
Also in the new version https://github.com/pi-hole/PADD/issues/162#issuecomment-877611854 ?

grug0r, 2021-07-10:
Yes.

DL6ER, 2021-07-10:
So one more question: What is the output of
```
ip addr | grep 'inet6 ' | grep -v '::1/128' | awk '{print $2}' | cut -f1 -d'/'
```

grug0r, 2021-07-10:
root@pihole:~# ip addr | grep 'inet6 ' | grep -v '::1/128' | awk '{print $2}' | cut -f1 -d'/'
fe80::b051:45ff:fec7:b165

[image]

DL6ER, 2021-07-10:
Weird, this is exactly what is happening in the original `padd.sh` but fixed in my proposed version, see the diff here https://github.com/pi-hole/PADD/pull/163

Can you double-/tripple check you use a `padd.sh` version with the line with the fixed version as shown [here](https://github.com/pi-hole/PADD/pull/163/files)?

grug0r, 2021-07-10:
Oh my apologies, don't  I feel dumb. When I was grabbing the fixed padd.sh it appended .1 i.e. padd.sh.1
Now it's showing..

[image]

DL6ER, 2021-07-10:
So the fix is confirmed working, thanks!

grug0r, 2021-07-10:
Yes, confirmed, real sorry for my stupidity. Many thanks.

DL6ER, 2021-07-11:
We just released v3.6.1 which contains the fix we tried here. Thanks for the report and enjoy!

christianboyle, 2021-07-11:
@DL6ER I was having this same issue and it appears v3.6.1 fixes it, but there are 2 oddities.

1. When PADD first starts up it briefly shows my IPv4 Wi-Fi address and then displays the IPv4 ethernet address.
2. In addition to this I noticed PADD now displays a `+` after the IPv4 ethernet address (rather than a `/24` as in previous versions). You can see this is @grug0r's screenshot above.

Is the `+` meant to indicate that there is more than one IPv4 address, but we only see one of them due to screen size limits?

DL6ER, 2021-07-12:
> Is the + meant to indicate that there is more than one IPv4 address, but we only see one of them due to screen size limits?

Yes, exactly this. Which address is shown is not specifically selected, it should be the first one returned by `ip addr`.

The CIDR netmask is omitted to give space for the `+`. Showing the netmask should not have any real valuable meaning in this place and may rather confuse users which are not aware of what this means (the plus seems to be similar now ;-)