pi-hole/PADD #172: CPU Load on RasPi Zero hovering at 110%-140%

ASchneider-GitHub, 2021-08-23:
**Describe the bug**
Downloading the most recent version of `padd.sh` from the GitHub and running it on a RasPi v4 with PiHole shows the CPU load consistently over 100% and rarely sinks below it.

**To Reproduce**
Steps to reproduce the behavior:
1. `wget https://raw.githubusercontent.com/pi-hole/PADD/master/padd.sh`
2. `chmod +x ./padd.sh`
3. `./padd.sh`

**Expected behavior**
"CPU Load" should be somewhat close to what `htop` is seeing, which is much lower (although spikes every 5 seconds when PADD is running, which I assume is the data gathering method running)

**Screenshots**
[image]

ASchneider-GitHub, 2021-08-23:
This issue can be closed. I realize now that disabling `newrelic-infra` service and waiting a while slowly brings the `CPU Load` back down to 50ish%. Turns out it can't run the New Relic agent at the same time as PADD. It has to be one or the other for some reason.

dschaper, 2021-08-23:
CPU Load looks pretty close to your load averages?

ASchneider-GitHub, 2021-08-24:
Hey there. This issue can be closed. I misinterpreted the readout of `htop` and thought the totals were way off. My bad!

closed 2021-08-24 by ASchneider-GitHub