pi-hole/PADD #174: piHoleVersion: line 9: unexpected EOF while looking for matching `)'

captcarl13, 2021-09-29:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
piHoleVersion: line 9: unexpected EOF while looking for matching `)'

**To Reproduce**
Steps to reproduce the behavior:
1. Update pi-hole to  core 5.5, web 5.7, FTL 5.10
2. Run PADD

**Expected behavior**
PADD expected to run with no error.

**Screenshots**
[image]

**Additional context**
On PADD v3.6.2 (current release)

captcarl13, 2021-09-30:
Updated FTL again to 5.10.1 and issue appears resolved.

rowlesmr, 2021-10-01:
I have the exact same problem with PADDmini

Pi-hole v5.5
AdminLTE v5.7
FLT v5.10.2

captcarl13, 2021-10-01:
I just updated and I'm not getting the same error, but version numbers are not appearing.
[image]
[image]

rowlesmr, 2021-10-01:
There is a fix here https://github.com/pi-hole/PADD/issues/175#issuecomment-931966859

captcarl13, 2021-10-01:
#175 fixed my issue with `pihole -r.` Marking as closed.

closed 2021-10-01 by captcarl13