pi-hole/PADD #175: PADD says updates available when they are not

JansonBHong, 2021-09-30:
I am running PI-Hole on a Raspberry Pi 4b with a full updated Buster. I have the latest PADD and all was working well until I updated Pi-Hole to the latest version today. PADD says Core and FTL are up to date but that updates are still available. The last line displayed on the screen  or the PADD display is

SYSTEM ========================================================================
 Uptime:   0 days, 2 hours, 54 minutes             Memory:   [■·········] 13.2% 
 CPU Temp: 43.3°C    CPU Load: 1.31, 0.86, 0.67    CPU Load: [■■■■■■■■■■] 131.0%piHoleVersion: line 9: unexpected EOF while looking for matching `)'

surtarso, 2021-10-01:
piHoleVersion: line 9: unexpected EOF while looking for matching `)'

gettin that same line in the end here

snoopyh42, 2021-10-01:
Seeing the same on my system.

enriqueardieanbudiono, 2021-10-01:
same problem here

enriqueardieanbudiono, 2021-10-01:
I got a temporary solution:
- type vim ./piHoleVersion
- go to ftl_version=
- fill the version number
- go to ftl_version_latest
- fill it with the same version number

NOTE:

This is only temporary, if you restart the pi, it will show the error again

kagegoomba, 2021-10-01:
Seeing the same issue here.
Tried the trick that enriqueardieanbudiono - nice temp fix that makes the screen look decent. I'll watch this thread for updates. ^_^

MrClever, 2021-10-01:
Looks like `pihole -v -f` spits out additional words which is breaking the indexes in `padd.sh`.

Fixed with this patch
```bash
--- /GitHubSource/PADD/padd.sh 2021-09-14 21:46:40.605798260 +1000
+++ padd.sh 2021-10-01 17:04:07.366645991 +1000
@@ -527,8 +527,8 @@
 
     # Gather FTL version information...
     read -r -a ftl_versions <<< "$(pihole -v -f)"
-    ftl_version=$(echo "${ftl_versions[3]}" | tr -d '\r\n[:alpha:]')
-    ftl_version_latest=${ftl_versions[5]//)}
+    ftl_version=$(echo "${ftl_versions[4]}" | tr -d '\r\n[:alpha:]')
+    ftl_version_latest=${ftl_versions[6]//)}
     if [[ "${ftl_version_latest}" == "ERROR" ]]; then
       ftl_version_heatmap=${yellow_text}
     else
```

JansonBHong, 2021-10-01:
Working perfectly - many thanks

yubiuser, 2021-10-01:
@MrClever 

Can you file a pull request to fix this?

MrClever, 2021-10-01:
Sure - will need to do the whole branch and fix dance. Give me a few minutes 👍🏻

kagegoomba, 2021-10-01:
Can confirm the hotfix works - I edited the file manually myself as I need to get to bed - but I can at least say it fixes the issue. :)
Much thanks!

MrClever, 2021-10-01:
PR #176 created :)

yubiuser, 2021-10-01:
The reason for the issue has been identified. Pi-hole switched from CI to GHA to build the binaries. This introduced an additionally `release` in the versions output. Please run `pihole -r` to fetch the new builds which should have corrected this.

MrClever, 2021-10-01:
Ran `pihole -r` and just did a repair. Then ran `pihole -v -f` again and it's back to the original output, so this issue goes away. Without the need for PR #176 

```
pihole -v -f
  FTL version is v5.10.2 (Latest: v5.10.2)
```

yubiuser, 2021-10-01:
Thanks for confirming :)

MrClever, 2021-10-01:
For the sake of absolute clarity, the current `padd.sh` release (v3.6.2) works without modification after doing the repair (`pihole -r`) and the changes proposed in #176 are not required..

captcarl13, 2021-10-01:
Can confirm repair with `pihole -r` corrects issue, making my issue #174 irrelevant.

kagegoomba, 2021-10-02:
Confirming that doing a pihole -r fixed the issue and after restoration of padd.sh

Nice to see a helpful community. :) Much thanks everyone.

closed 2022-06-28 by yubiuser