pi-hole/PADD #177: IP has a random + after it

SodaWithoutSparkles, 2021-10-02:
**Describe the bug**
When running padd.sh, the IP has a + sign following it, even after `pihole -r` repair.

**To Reproduce**
Steps to reproduce the behavior:
1. `./padd.sh`
2. Observe the IP has a + sign following it, for example: `IP:       192.168.1.107+`

**Expected behavior**
`IP:       192.168.1.107`

**Screenshots**
[image]

**Additional context**
```
pi@raspberrypi:~ $ pihole -v
  Pi-hole version is v5.5 (Latest: v5.5)
  AdminLTE version is v5.7 (Latest: v5.7)
  FTL version is v5.10.2 (Latest: v5.10.2)
```

Also I noticed that there are two numbers for the FTL PID, is that normal?

kagegoomba, 2021-10-02:
I think the + means there's more than one IP Address - WiFi Card active? - Saw something about this elsewhere.

SodaWithoutSparkles, 2021-10-02:
```
pi@raspberrypi:~ $ sudo ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Half 1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Half 1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: Unknown
        Supports Wake-on: gsf
        Wake-on: d
        SecureOn password: 00:00:00:00:00:00
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
pi@raspberrypi:~ $ sudo ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.107  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::3f33:e9e:4994:437a  prefixlen 64  scopeid 0x20<link>
        ether REDACTED  txqueuelen 1000  (Ethernet)
        RX packets 10349402  bytes 1639316777 (1.5 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8196679  bytes 2135489324 (1.9 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 632305  bytes 75208221 (71.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 632305  bytes 75208221 (71.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 6000
        inet 10.8.0.1  netmask 255.255.255.0  destination 10.8.0.1
        inet6 fe80::57a4:7eb1:8ae0:7aa2  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 38672  bytes 5128676 (4.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 82104  bytes 78256622 (74.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Wifi was blocked by rfkill

kagegoomba, 2021-10-02:
tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 6000
        inet 10.8.0.1  netmask 255.255.255.0  destination 10.8.0.1   <---- I suspect this is your 2nd IP (tun0)
I just presumed your wifi card was the source.
Again - not sure..but I'm fairly certain that PADD will use a + if more than one IP is "up" on the device.

SodaWithoutSparkles, 2021-10-03:
Humm, I think that was the ovpn IP, that explains it. I would change the title.

Also, why are there two FTL PID?

closed 2021-10-03 by SodaWithoutSparkles

kagegoomba, 2021-10-03:
Not sure - I only see one on mine. I'll have to let someone else speak to that.

DL6ER, 2021-10-03:
More than one PID might happen when dedicated workers are created. This is 100% okay and happens if:

1. At least one client is asking DNS queries of TCP instead of UDP (each TCP client gets its own dedicated worker)
2. You are running scripts (e.g. to post-process DHCP events), they are handled by a dedicated worker as well