pi-hole/PADD #178: PADD displays error message about "invalid number"

sa0azs, 2021-11-14:
Freshly installed Pi-Hole and PADD on a Raspberry PI 2B and on a Zero W, both with new installs of Raspberry PiOS Bullseye. 

**Describe the bug**
Shown during startup and blinking on screen while running, are the following error messages: 

/home/pi/padd.sh: line 900: printf: 0,3: invalid number
/home/pi/padd.sh: line 244: printf: 42,236: invalid number
/home/pi/padd.sh: line 873: printf: 9,1: invalid number
/home/pi/padd.sh: line 900: printf: 0,91: invalid number

DL6ER, 2021-11-14:
Did you configured a different locale (not English, UK)?

pheiduck, 2021-11-14:
go this too: (locale is DE only)

Log Trace:
```
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8683: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8357: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8289: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8221: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8237: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8169: invalid number
/opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
/opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
/opt/pihole/chronometer.sh: line 398: printf: 30,8254: invalid number
```

sa0azs, 2021-11-16:
> 
> 
> Did you configured a different locale (not English, UK)?

Choosen locale is sv-SE, but en-GB (the default) is also installed.

PADD worked fine with RaspberryPiOS Buster and the same locale settings on both PI:s, so I wonder if there is something changed in Bullseye that causes this?

pheiduck, 2021-11-29:
> go this too: (locale is DE only)
> 
> Only happend on Raspberry Pi OS 11 (10 has not this issue
> 
> Log Trace:
> 
> ```
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8683: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8357: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8289: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8221: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8237: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8169: invalid number
> /opt/pihole/chronometer.sh: line 315: printf: 1,4: invalid number
> /opt/pihole/chronometer.sh: line 322: printf: 55,844: invalid number
> /opt/pihole/chronometer.sh: line 398: printf: 30,8254: invalid number
> ```

Fixed on next release:
https://github.com/pi-hole/pi-hole/pull/4085

pheiduck, 2021-12-22:
Still present on latest Pi-hole updates…

pheiduck, 2021-12-30:
I think its because of python 3:
here: https://github.com/pi-hole/pi-hole/blob/1b809e4e8ea442cd2a3cc54206429fabd58cce05/advanced/Scripts/chronometer.sh#L315

It should be
```
[[ "$cpu_mhz" -le "999" ]] && cpu_freq="$cpu_mhz MHz" || cpu_freq="$(printf ("%.1f" $(calcFunc "$cpu_mhz"/1000))) GHz"
```
Or
```
[[ "$cpu_mhz" -le "999" ]] && cpu_freq="$cpu_mhz MHz" || cpu_freq="$(printf ("%.1f") $(calcFunc "$cpu_mhz"/1000)) GHz"
```

dschaper, 2021-12-30:
PADD doesn't use Python, it's all bash.

pheiduck, 2021-12-30:
> Did you configured a different locale (not English, UK)?

okay change the locale to "en_GB.UTF-8" does the trick.
`export LC_ALL="en_US.UTF-8"`
Language doesn't matter on a server XD

pheiduck, 2021-12-31:
A similar error was reported here. https://github.com/jpmck/PADD/issues/5

closed 2022-12-17 by github-actions[bot]