pi-hole/PADD #179: Add CPU Temp to PADDmini display

dynamix911, 2021-11-19:
Hello,

Can CPU Temp be added to the PADDmini display. I have a 2.8" adafruit display, and there is plenty of room at the bottom to add CPU Temp. I have tried to modify the padd.sh file to add it, and did not work.

Also the display will show the PADDmini display and every min or so it will show a sort of what looks like a help screen and then jumps back to the regular display.

Thank You

dynamix911, 2021-11-21:
Buy running sudo dpkg-reconfigure console-setup, and changing the font to 6x12. My whole screen gets filled with all the info and looks great. It switches to PADD Tiny from PADD Mini. How can i keep it at this size? on reboot it reverts back to the mini size and then i have the lower half of the screen empty. Have to run a sudo dpkg-reconfigure console-setup and change font to 6x12.

I also have this strange screen that pops up every 30 sec or so
[image]

yubiuser, 2022-08-20:
For "mini" to be active, the screen needs at least 40x18 (columns x lines). But then there is no space left to include CPU temp.
[image]

The only option I see is shrinking the load bar and placing it on the right. But this would render the bar pretty meaningless....

closed 2022-12-17 by github-actions[bot]