pi-hole/PADD #180: Strat problem PADD

Falconcrest, 2021-12-17:
Hi there,

First of all, PADD is a nice tool for the Pi-hole.
I had to rebuild my system after a crash.
It consists of the RPi4 4GB, an SSD, DietPi 7.9 Armv7 32-bit image (Debian), Pi-hole, Unbound and a 3.5 "display.
I had to adjust the software for the display a little, change / home / pi to / root twice, then the installation worked.
PADD now works almost flawlessly, there is only one temporary problem. When the system starts, a message appears on the display where a few conditions are queried.
 
START UP
- Checking internet connection ...
OK start: lighttpd
OK start: cron

... and now appear one after the other

Attempt 1 failed x seconds
Attempt 2 failed x seconds
until
Attempt 9 failed 128 seconds.

It is counted down to 0 and now PADD starts without errors and works 24/7. What could be the problem?

Best regards, Carsten

Jarsky, 2021-12-24:
Mines started doing the same thing recently. Only thing thats changed is regular Ubuntu updates and Pi-hole update

yubiuser, 2021-12-27:
Please check if `netcat` is installed.

Falconcrest, 2021-12-27:
That was it. Thanks to all involved.

closed 2021-12-27 by Falconcrest