pi-hole/PADD #181: Netcat is required but requirement not documented

yubiuser, 2021-12-27:
PADD uses netcat to check for internet connection.

https://github.com/pi-hole/PADD/blob/00ec49d720974e27d935946f13c6410867f38f0b/padd.sh#L970

But there is no check if netcat is installed at all nor is it listed as requirement. When netcat is not present, the function will fail 9 times before the script continues. See this issue: https://github.com/pi-hole/PADD/issues/180

Netcat was once a Pi-hole requirement and subsequently installed by default, but not longer necessary. It was removed by:
https://github.com/pi-hole/pi-hole/pull/4346

PromoFaux, 2021-12-28:
Could this usage be replaced with `ping`? In much the way we do in the debug script? I think `ping` installed as standard in most distros?

yubiuser, 2021-12-30:
`netcat` has been (for other reasons) re-added to `pihole` 
https://github.com/pi-hole/pi-hole/pull/4485/commits/2a869419b4d7b958f98f685be8a96f4da7c594e4

innovarew, 2022-01-02:
Great!, the same change 2a86941 could be replicated in docker-pi-hole so PADD works out of the box.

DL6ER, 2022-01-02:
docker-pi-hole will incorporate the changes at the next official Pi-hole release where the changes currently in `development` get released. This does not necessarily have to be the next docker image, but it'll come soon.

PromoFaux, 2022-01-02:
The next docker image will have it even if we tag that before the next Pi-hole release - as I've added it to the base image:

https://github.com/pi-hole/docker-base-images/commit/0b12aee440e72182a5d6cc622aa3610b920df110

(having these base images with all the dependencies on them makes things much quicker when building the final image!)

Should be tagging `2022.01` for docker later on today.. (maybe)

thomasmerz, 2022-02-02:
@PromoFaux , @yubiuser 
This could easily be fixed by this which works fine with every `bash`:
```
>/dev/tcp/hostname/portnumber
```

DL6ER, 2022-02-02:
You mean like https://github.com/pi-hole/PADD/blob/1ce1d339a425a06d744009c40a4b7780784c8230/padd.sh#L124 ;-)

yubiuser, 2022-02-02:
I think this issue can be closed because netcat is again a Pi-hole dependency

https://github.com/pi-hole/pi-hole/blob/5d68dac90eafe2fd2eb9693507acc1911a048050/automated%20install/basic-install.sh#L290

closed 2022-02-02 by yubiuser