pi-hole/PADD #183: FTL v5.13 and Core v5.8 broken the connection to FTL

giuliomagnifico, 2022-01-06:
**Describe the bug**
Latest FTL v5.13 and Core v5.8 update has changed the way FTL checks and gets the datas. Here more infos: https://github.com/pi-hole/pi-hole/pull/4485/commits/f45248df805c426f9d9e654c0f71eb1fd8b795e8 

The results is unknow status, changes must be made here:

```
GetFTLData() {
  local ftl_port LINE
     ftl_port=$(cat /run/pihole-FTL.port 2> /dev/null)
     if [[ -n "$ftl_port" ]]; then
         # Open connection to FTL
         exec 3<>"/dev/tcp/127.0.0.1/$ftl_port"

         # Test if connection is open
         if { "true" >&3; } 2> /dev/null; then
             # Send command to FTL and ask to quit when finished
             echo -e ">$1 >quit" >&3

             # Read input until we received an empty string and the connection is
             # closed
             read -r -t 1 LINE <&3
             until [[ -z "${LINE}" ]] && [[ ! -t 3 ]]; do
                 echo "$LINE" >&1
                 read -r -t 1 LINE <&3
             done

             # Close connection
             exec 3>&-
             exec 3<&-
         fi
     else
         echo "0"
     fi

```

**Screenshots**
[image]

yubiuser, 2022-01-06:
This issue is fixed by
https://github.com/pi-hole/PADD/pull/182

But you found a place that could be improved 😉

A12BA5E, 2022-01-06:
I confirm same issue.

giuliomagnifico, 2022-01-06:
@A12BA5E change the line 422 https://github.com/pi-hole/PADD/pull/182/files with the updated line and it will work again

[image]
 
@yubiuser thanks for the reply, i fixed it.

A12BA5E, 2022-01-06:
@giuliomagnifico Did the trick! Thank you!

yubiuser, 2022-06-28:
This was fixed by https://github.com/pi-hole/PADD/pull/182

closed 2022-06-28 by yubiuser