pi-hole/PADD #187: External Address for Pi-hole

EndreEndi, 2022-01-19:
So, i want to install PADD onto a Linux which hasn't Pi-hole installed, i have Pi-hole installed in a docker, what i can do to connect the PADD script to the Pi-hole without them being onto the same machine.
Sorry for bad English.

thomasmerz, 2022-02-19:
You should copy `padd.sh` to your linux-with-pihole in this (the first) directory `/home/thomas/dev/github/docker-pi-hole/etc-pihole` so it's accessible inside docker container:
```
$ docker inspect pihole | grep /etc/pihole
                "/home/thomas/dev/github/docker-pi-hole/etc-pihole:/etc/pihole:rw"
                "Destination": "/etc/pihole",
                "/etc/pihole": {}
```

Then you can do this (assuming that you are able to ssh to your linux-with-pihole):
```
$ docker -H ssh://root@linux-with-pihole exec -it pihole pihole -v
$ docker -H ssh://root@linux-with-pihole exec -it pihole /etc/pihole/padd.sh
```

This shows up to this:

[image]

thomasmerz, 2022-02-28:
@EndreEndi , does this solve your problem and answer your question? If this is the case, this issue should be closed. See also my [comment](https://github.com/pi-hole/PADD/pull/155#issuecomment-1045327841) in PR #155.

thomasmerz, 2022-03-18:
I think I provided a solution for this issue. Can someone close this please? Thanks a lot 👍🏻

EndreEndi, 2022-03-19:
Hi, sorry for the long time response, i don't really check Github a lot, i will close it for now, and try later the solution, Thank you for helping if works, if not, thanks for trying :)

EndreEndi, 2022-03-19:
Hi again, i 'm a bit stuck, my container name is pihole-template
[image]
i was trying to find where to put the name of the container

EndreEndi, 2022-03-19:
I made it work, sorry for open / close spam, Thank you.
[image]

closed 2022-03-19 by EndreEndi

thomasmerz, 2022-03-19:
It was me a pleasure to help you, @EndreEndi 😃