pi-hole/PADD #192: PADD 3.6.4 Shows as out of date

DesktopECHO, 2022-02-07:
Looks like you need formalize release 3.6.4 on GitHub.  
Right now latest is showing 3.6.3 but PADD uses release tags to figure out if it's current.

``padd_version_latest=$(json_extract tag_name "$(curl -s 'https://api.github.com/repos/pi-hole/PADD/releases/latest' 2> /dev/null)")``

[image]

[image]

dschaper, 2022-02-07:
Thanks, I just tagged that for a release. Should see it correctly now.

DesktopECHO, 2022-02-07:
Looks good, thank you kindly!

closed 2022-02-07 by DesktopECHO