pi-hole/PADD #200: Increase PADD to fill the whole screen

N0Klu3, 2022-03-02:
**Is your feature request related to a problem? Please describe.**
I have a 7" screen running 1024 x 600.

I want to use this as a screen to always run on the side of my office shelf as a visual aid.
But its a bit small as it is. 

**Describe the solution you'd like**
Is there any way to increase the size to fill the screen so I can read it from afar easier.

**Describe alternatives you've considered**
N/A

**Additional context**
[image]

socrateslopes, 2022-07-11:
Hello, @N0Klu3
Have you tried this https://github.com/pi-hole/PADD/wiki/Sizes ?
Please let me know if it helps.

XiphiqiX, 2022-07-28:
Hi @N0Klu3 

I also have this size screen and the same issue.

When using sudo dpkg-reconfigure console-setup, none of the available font sizes scale correctly to the 7inch screen.
I was using 10x18 (framebuffer  only) and this worked in PADD 3.7

I'm currently using 8x16 and this at least allows all information in the script to be displayed, but does not fit the screen properly

[image]
[image]
.

yubiuser, 2023-01-01:
Starting with PADD v3.9.0. (https://github.com/pi-hole/PADD/releases/tag/v3.9.0) we use a centered layout by default. This should improve the visual appearance.  That said it's unlikely that this request will be "really fixed" soon (or at anytime), because we would need to design a new layout for a bigger screen. PADD is a shell script, there is no easy way to style things or make the interface responsible to size. We solved the problem by having a few "fixed" layouts what will be shown depending on the screen size. As your screen is bigger than the biggest internal dashboard, you see a lot of empty space around it. We would need a new layout, with bigger distances between each dashboard item - converting the empty outer space to empty inner space.
However, pull requests are welcomed.

closed 2023-02-06 by github-actions[bot]