pi-hole/PADD #201: Port PADD to a new and more robust language.

dschaper, 2022-03-04:
**Is your feature request related to a problem? Please describe.**

`bash` is great but it's close to hitting it's limits as far as extensibility and maintenance. I'm considering porting this to a new language and I'm looking for thoughts and ideas. My first thought is to use `golang` but that is a compiled language and would remove some of the transparency of the code. All the code would still be open source and available here with instructions to compile yourself if you felt that was more comfortable.

I may turn this in to a discussion topic if there are people interested in this or have other ideas and concerns.

PromoFaux, 2022-03-04:
Was recently recommended these Go libraries for console stuff by @goblinfactory

- https://github.com/jroimartin/gocui
- https://pterm.sh/#/?id=%F0%9F%A7%AA-examples
- https://github.com/rivo/tview

May be of use here

dschaper, 2022-03-04:
Thanks, I was going through a few tutorials with 

https://github.com/spf13/cobra
and
https://github.com/manifoldco/promptui

goblinfactory, 2022-03-05:
Go is a great language if you've come from C# ; you'll enjoy the simplicity and it will make you a better C# programmer. 

Adam mentioned me by name, and this is the first time I've seen the code, so not familiar with the project history, just reading from the top, and then a quick peek at the project, (appears to be a single bash file  for self evident simplicity and reviewability and trusty-ness) ,that makes perfect sense.

I think the concern about transparency is valid. While go binaries can be distributed as Brew binaries; if you already have Go installed, you can install utilities from source with a single command, `go install {repo_url}` and that installs it from source, so for all the go users that will run a future PADD written go, they'd most likely be doing it that way. (for security reasons). 

my 2 cents;
Having taken a quick look at the bash script;  running off a single config,  I think the std lib `flags` package should be sufficient to get started with instead of cobra; and extra features like using `Prompt UI` might be better as another project that wraps the command line project, i.e just as a way to `wizard` config it "for you", thus keeping the core package size small for those who are happy to edit a config manually.

Here are some go references that I think would be relevant here;
*

dschaper, 2022-03-05:
Great! Thanks a ton for the helpful information. I've used the stdlib `flags` and it seemed a little limited. The `go-flags` may be helpful but thinking over things, I honestly don't see a whole lot of CLI arguments that need to be passed. 

`PADD` essentially queries `pihole-FTL`s telnet API for status and some basic metrics and presents it in a dashboard like fashion. 

And thank you for the idiomatic Go advice on additional libraries!