pi-hole/PADD #207: wrong permissions showing up

un99known99, 2022-03-25:
START UP ===================================================
- Checking internet connection...
  - Attempt 1 passed...
  - Connectivity check passed...
- Writing PID (22042) to file...
/usr/local/bin/padd.sh: line 1144: ./PADD.pid: Permission denied
- Checking for PADD version file...
  - PADD version file found... deleting.
rm: cannot remove 'piHoleVersion': Permission denied

padd.sh is "-rwxr-xr-x  1 root root 48738 Mar 25 10:30 padd.sh"

yubiuser, 2022-03-25:
Since PR https://github.com/pi-hole/PADD/pull/168 `padd` changes into the directory where `padd.sh` is located before it created `piHoleVersion` and `PADD.pid`.

Please check write permission of the user executing `padd.sh` and the directory it is located. 

P.S. I'm aiming to remove `PADD.pid` in the future anyways https://github.com/pi-hole/PADD/pull/197

un99known99, 2022-03-25:
Shouldnt be a problem?

[image]

[image]

yubiuser, 2022-03-25:
If you read the discussion of https://github.com/pi-hole/PADD/pull/168 you can see that we discussed of moving the files to `/tmp`. But we dropped it. The OP was convinced that 

> I think that it is much easier to simply instruct users to keep padd.sh in a writeable directory. And also, in almost all cases, the directory is going to be writable anyway (since padd.sh was created in that directory!).

I think I'm going to file a PR to move the files to `/tmp`

un99known99, 2022-03-25:
yes, because of the 2 tmp file being created, saw it in $HOME/bin, for me its ok using the "private" bin dir, also fine if there is "nice" move of them to /tmp, for padd.sh it is ok (at least for me ;) ), but an idea would also to add ar remove statedment in padd.sh once terminated to remove the 2 temp files ----  thx for ur comments :)

yubiuser, 2022-03-25:
https://github.com/pi-hole/PADD/pull/211 should fix the issues

yubiuser, 2022-05-23:
Should be fixed not with PADD 3.7.0