pi-hole/PADD #214: PADD wont update

dutchschaefer, 2022-04-18:
pihole version - v5.9.1 web v5.11. ftl v5.14

**Describe the bug**
Have tried a few different ways to update PADD but after rebooting it still shows as updates are available.

**To Reproduce**
Steps to reproduce the behavior:
Tried: cd ~
wget -N https://raw.githubusercontent.com/pi-hole/PADD/master/padd.sh
Also
pihole -up
also
pihole -r among a few other options that I can't remember over the past few days.

**Expected behavior**
pi-hole reboots but still shows the same error:  PADD v3.6.7. [i] updates are available

Falconcrest, 2022-04-18:
Unfortunately I have to confirm this behavior.

dschaper, 2022-04-18:
Can you run a `head -n 30 padd.sh` on your system so we can see what version you have locally?

iamenderst, 2022-04-18:
Got the same here:
# VERSION
padd_version="v3.6.7"

dutchschaefer, 2022-04-18:
[image]

Falconcrest, 2022-04-18:
the same thing

# VERSION
padd_version="v3.6.7"

dschaper, 2022-04-18:
You have the latest version. I think what you are seeing is not that PADD has updates available but instead that Pi-hole has updates available. There was a glitch on GitHub that reset the AdminLTE v5.11 release and that has been fixed. It will take some time to reflect that fix or you can run `rm /etc/pihole/GitHubVersions` and force a new check.

Falconcrest, 2022-04-18:
That was the solution to the riddle! Thanks!

dutchschaefer, 2022-04-18:
[image]

I am not the greatest at coding so what am I doing wrong?

iamenderst, 2022-04-18:
> 

sudo rm /etc/pihole/GitHubVersions

dutchschaefer, 2022-04-18:
Thanks that worked!

closed 2022-04-18 by yubiuser