pi-hole/PADD #215: PADD doesn't show version information on top screen

spintike, 2022-05-21:
**Describe the bug**
Running PADD v3.6.7 on Raspberry Pi 3B+ (Raspbian Jessie with kernel 5.15.40-v7) with PiHole v5.10. It's being display on the attached 3,5" Touchscreen and working fine so far, lovely colors 🥇 
I've ruled out initial problems with the LOCALE but now another issue appears. PADD does not show the information for the Core, FTL and Web version.
After searching a bit in the code, I've presumably found the issue and narrowed it down to the following code lines:

`read -r -a core_versions <<< "$(pihole -v -p)"`
`core_version=$(echo "${core_versions[3]}" | tr -d '\r\n[:alpha:]')`
(...)

`read -r -a web_versions <<< "$(pihole -v -a)"`
`web_version=$(echo "${web_versions[3]}" | tr -d '\r\n[:alpha:]')`
(...)

`read -r -a ftl_versions <<< "$(pihole -v -f)"`
`ftl_version=$(echo "${ftl_versions[3]}" | tr -d '\r\n[:alpha:]')`

When manually executing each command individually and querying the variables afterwards both variables are empty.

What I've found so far is:

- when manually executing `pihole -v -p` etc. the result is the following:

_Current Pi-hole version is v5.10_

- the `read` command does not need "-r -a" in order to work (with the parameters the variable `core_versions `contains _Current_ only)

- the string above has 5 parts, meaning `${ftl_versions[3]}` would return "is" and `${ftl_versions[4]}` would return "v5.10".

- the piped command `tr -d '\r\n[:alpha:]'` deletes everything except for digits.

So I have changed the code to the following, after that the variable `core_version` contains 5.10 as expected.

`read core_versions <<< "$(pihole -v -p)"`
`core_version=$(echo "${core_versions[4]}" | tr -d '\r\n[:alpha:]')`

**Expected behavior**
That the version information is shown normally.

**Screenshots**
[image]

[image]

It's possible that I am totally on the wrong way with it; I am hoping for an idea how to solve this!

Thanks in advance for your consideration!

yubiuser, 2022-05-21:
Thanks for your detailed analysis. I have reworked the whole script to make it more POSIX compliant and the lines you mentioned have been changes as well. The code has not been shipped yet, but is part of the `development` branch. You can find it here, feel free to try it and give feedback.

https://github.com/pi-hole/PADD/blob/development/padd.sh

spintike, 2022-05-22:
> Thanks for your detailed analysis. I have reworked the whole script to make it more POSIX compliant and the lines you mentioned have been changes as well. The code has not been shipped yet, but is [...]

Thanks a lot, really appreciate it! Now indeed the display works just fine:

[image]

closed 2022-05-22 by spintike