pi-hole/PADD #224: Illegal number: inf / Pi-Hole versions not showing

Yaff1e, 2022-05-26:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**

v3.7.0 and 3.7.1 show the errors below when loading. If loading completes it then shows v in all the Pi-Hole versions.

PADD version file not found

./padd.sh: 882: [: Illegal number: inf
./padd.sh: 884: [: Illegal number: inf
./padd.sh: 882: [: Illegal number: inf
./padd.sh: 884: [: Illegal number: inf
./padd.sh: 882: [: Illegal number: inf
./padd.sh: 884: [: Illegal number: inf

**To Reproduce**
Steps to reproduce the behavior:

Load PADD

**Expected behavior**

PADD to load and show version numbers

**Screenshots**

[image]

[image]

**Additional context**

I updated to v3.7.0 yesterday and 3.7.1 today on both of my Pi-Holes. One of them is fine but the other not. It shows the errors above when loading. It then sits on that screen for a long time and either doesn't go past it or if it does it shows v in all the Pi-Hole versions with the same errors at the bottom of the screen.

yubiuser, 2022-05-26:
What's the output of `pihole -v`

Yaff1e, 2022-05-26:
[image]

yubiuser, 2022-05-26:
The version numbers are very likely a red hering. The issue is

https://github.com/pi-hole/PADD/blob/4c888a894bb4c7e2b4a4526c17c21b1b9b14d31e/padd.sh#L240

returning `inf` (infinity).

What is your output of `cat /sys/devices/system/cpu/kernel_max`

Yaff1e, 2022-05-26:
It returns 0

yubiuser, 2022-05-26:
@Yaff1e 

We discovered the bug. Please try if this version is fixing the issue for you:
https://github.com/pi-hole/PADD/blob/fix/core_count/padd.sh

Yaff1e, 2022-05-26:
Yes, that's fixed it thanks.

Yaff1e, 2022-05-27:
That returns 1

yubiuser, 2022-07-30:
Fixed with https://github.com/pi-hole/PADD/releases/tag/v3.8.0

closed 2022-07-30 by yubiuser