pi-hole/PADD #246: Unexpected color changes in newest update

bctstkla, 2022-07-17:
Hi,

after updating padd there are unexpected color changes that do not look good at all.
Was there some reason for this change and how do I revert the colors to the original ones?
(Other than running an older version of padd from now on.)

[image]

yubiuser, 2022-07-17:
We did not change any colors. Maybe your `sh` uses a different color schema than `bash`. Please try to change the first line in the script from `#!/usr/bin/env sh` to `#!/usr/bin/env bash`

and see if this resolves your issue.

bctstkla, 2022-07-17:
Hi @yubiuser and thanks for your suggestion, the line in question does indeed differ between the old and new version of padd.sh. Unfortunately modifying the file only changes the behavior (colors) when connected to the Pi with PuTTY and running PADD, but running it on the LCD of the Pi still has these weird colors. Even if I run padd.sh manually on the Pi by hooking up a keyboard.

bctstkla, 2022-07-17:
I ran a text comparison between the working an non-working version, and the culprit is changes in the color definitions (below).
[image]

I reverted these changes, and now the color are back to the expected way.

yubiuser, 2022-07-17:
The change was made by PR https://github.com/pi-hole/PADD/pull/171

dschaper, 2022-07-18:
Can you get the term type for your setup?

`echo $TERM`

bctstkla, 2022-07-25:
Hi @dschaper,
running directly from raspberry's terminal (with the lcd) the term type is "linux" and the color outputted is that weird neon-white-yellowish.

Running via ssh and putty, term type is xterm and the color is the expected (green).

dschaper, 2022-07-25:
Thanks, it seems that the issue is the terminal type. I don't remember offhand how to set the `tty`s to a term that would render the correct escape codes. It may be in the default system `.profile`?

closed 2022-12-17 by github-actions[bot]