pi-hole/PADD #251: PADD 3.8.0 display error with 3.5" PiTFT

Falconcrest, 2022-07-27:
After the update to padd 3.8.0 at least the first line above with the details of the versions of pihole, web and padd is missing.

yubiuser, 2022-07-27:
Is there some real error message shown? Or only the misalignment? Could you add a screenshot or photo?

Falconcrest, 2022-07-27:
only the misalignment without real error message.

cmdshft, 2022-07-27:
Same issue here with a Hyperpixel 4 display

[image]

yubiuser, 2022-07-27:
Thanks I think I know where this is coming from. It's probably due to the two new lines adding model and inteface.

Falconcrest, 2022-07-27:
Maybe you can correct something there so that it fits again.

yubiuser, 2022-07-27:
What do you want me to remove instead?

yubiuser, 2022-07-27:
Please run `stty size` and post the output. I need to see how big the terminal is.

yubiuser, 2022-07-27:
Everything above 79x25 is considered `mega` so far and assumed there are enough lines
https://github.com/pi-hole/PADD/blob/380636a96ac5a7f76291d994b901879709b9af4f/padd.sh#L926

yubiuser, 2022-07-27:
[image]

26 - 27 - 28 in a row. I think we should go for 28.

yubiuser, 2022-07-27:
`tiny` and `mini` are broken as well.

Falconcrest, 2022-07-27:
root@pihole:~# stty size
59 183

yubiuser, 2022-07-27:
And `tput cols` and `tput lines`

Falconcrest, 2022-07-27:
root@pihole:~# tput cols
183
root@pihole:~# tput lines
59

yubiuser, 2022-07-27:
But then it should easily fit...

https://github.com/pi-hole/PADD/blob/380636a96ac5a7f76291d994b901879709b9af4f/padd.sh#L1108-L1109

So we get `console_width=183` and `console_height=59` wich passes 
https://github.com/pi-hole/PADD/blob/380636a96ac5a7f76291d994b901879709b9af4f/padd.sh#L926-L934

And we should get size `mega`

Falconcrest, 2022-07-27:
I compared it, the code is identical, but what can I say, 1 line is missing at the top.

Falconcrest, 2022-07-27:
Is there a solution? Thanks a lot for your efforts!

yubiuser, 2022-07-27:
I haven't figured out one yet. I don't know why it does not fit, although there are enough lines. The trivial solution would be to remove newly added lines. In the next few days I won't have time to work on this. In the meantime you could just use the previous version.

Falconcrest, 2022-07-27:
OK, we will wait, thanks!
Sorry, by the way there are 2 lines that are not displayed when I look very closely...26 lines are displayed in version 3.7.1.

nineteenletterslong, 2022-08-01:
I have a PiTFT 3.5" and have the exact same issue with the line being cut off.

`pi@pihole:~ $ tput cols
364
pi@pihole:~ $ tput lines
89
`

EdmundGerber, 2022-08-01:
> What do you want me to remove instead?

The logical things to remove would be the two new items. For now I'm sticking with 3.7.1, so same difference, for me.

nineteenletterslong, 2022-08-02:
Possibly split some information out a tab or two instead of having separate lines, if possible.

cmdshft, 2022-08-19:
fix coming soon?

yubiuser, 2022-08-19:
@Falconcrest 

> root@pihole:# tput cols
183
root@pihole:# tput lines
59

@nineteenletterslong 

>pi@pihole:~ $ tput cols
364
pi@pihole:~ $ tput lines
89
___

I spend some thoughts on this today and I think I know why the numbers are so high: did you run those commands in a terminal connected via `ssh` to your RPi? They certainly reflect the size of your terminal on your device accessing the Pi, but not the terminal displayed at your RPi display. Could you run those commands `directly` on the Pi?

nineteenletterslong, 2022-08-19:
Great point!

Just plugged a keyboard in and ran it on the pi (again, PiTFT 3.5")

pi@pihole:~ $ tput cols
80
pi@pihole:~ $ tput lines
26

yubiuser, 2022-08-19:
Thanks, this looks reasonable.

Falconcrest, 2022-08-19:
I would confirm the information from @ninetennletterslong. 
In the absence of a currently unavailable keyboard, I counted. ;-)

yubiuser, 2022-08-20:
I fixed the issue. In "mini", IP is now shown next to hostname. In "tiny" I had to sacrifice the IPv6 address. In "regular" and "slim" I removed the device model and IPv6 address but added an IPv6 checkbox. In "mega" I combined DNS with the DHCP section.
[image]
[image]
[image]
[image]
[image]
 
 ___
 
 Testing and feedback welcome. You can find this version at
 
 https://raw.githubusercontent.com/pi-hole/PADD/0b4351c3adf2069d63a7b2c0cba890ab165802d1/padd.sh

Falconcrest, 2022-08-20:
It works! A very, very good job! Thank you, it was worth the wait.

nineteenletterslong, 2022-08-20:
Works great, still have red DNSSEC though but thanks for your efforts

rdwebdesign, 2022-08-21:
In small screens you can combine a shorter `uptime` text and `cpu temp` on a single line.

To get a shorter `uptime` you can use:
```sh
system_uptime=$(awk '{print int($1/86400)"d:"int(($1%86400)/3600)"h:"int(($1%3600)/60)"min"}' '/proc/uptime')
```
It will return something like this:
`5d:13h:41min`

cmdshft, 2022-08-21:
When will this be pushed to the main release link so I can update normally?

yubiuser, 2022-08-21:
It'll take a bit. The PR that changes that is still in draft mode as I'm still working on it.

EdmundGerber, 2022-09-02:
It's been 2 weeks - is that a normal amount of time for the changes you made?

yubiuser, 2022-09-02:
> It's been 2 weeks - is that a normal amount of time for the changes you made?

As we all do this in our free time - yes. You can already use the version I linked above.

iamenderst, 2022-09-05:
Your testing version fixed the issue for me. PiTFT Plus 480x320 3.5

v3.7.1
[image]

Release v3.8.0
[image]

Testing v3.8.0
[image]

yubiuser, 2022-09-05:
Thanks for the testing. I was involved a bit more in the latest changes of "core" Pi-hole but will focus now again on PADD. It's probably only a few hours to finish this change. With all this testing it should convince the reviewers easily.

accordxtc, 2022-09-12:
Thanks @yubiuser for your help fixing this. Just applied to my screen, missed having it fit all the correct data. Much appreciated

ashblade, 2022-09-20:
Thanks so much for all you do @yubiuser ! Fixed 3.8.0 version posted above works great on my PiTFT as well. Good to see that everything fits again. 🥇

cmdshft, 2022-09-30:
It's been 3 months, can the fixes be pushed to the main release yet?

yubiuser, 2022-09-30:
I was going to finish this but in the meantime Pi-hoel/ `FTL` had made some changes so I could not go on. We now first need to merge https://github.com/pi-hole/PADD/pull/260

iamenderst, 2022-09-30:
The testing version in the link above works fine until 3.8.1 is release.
Thanks yubiuser!

EdmundGerber, 2022-10-11:
> Testing and feedback welcome. You can find this version at
> 
> https://raw.githubusercontent.com/pi-hole/PADD/0b4351c3adf2069d63a7b2c0cba890ab165802d1/padd.sh
> 
> P.S. The red "DNSSEC" on "mega" is fixed already

I've been using the version of 3.8.0 that was linked, and it's been fine until yesterdays PiHole update ([Pi-hole FTL v5.18.2, Web v5.16 and Core v5.13). Padd starts, shows a red X for Core, and beside that states 'Pihole's DNS server is off!'. However, the web interface shows pihole is working as normal. I've tried Pihole -r, but no change. Any ideas?

yubiuser, 2022-10-11:
This is expected. PR https://github.com/pi-hole/PADD/pull/260 fixed this.
I'm waiting approval on https://github.com/pi-hole/PADD/pull/257 (which is the fix for this visual issue) and will immediately release a new PADD version with both fixes included.

EdmundGerber, 2022-10-11:
Thanks for the reply. Knowing it's expected behaviour - and not something hamfisted I did - is a plus.

yubiuser, 2022-10-12:
Finally fixed with https://github.com/pi-hole/PADD/releases/tag/v3.8.1

closed 2022-10-12 by yubiuser