pi-hole/PADD #261: Reporting as "DNS Offline" after Pihole upgrade

rhorrell, 2022-10-11:
**Describe the bug**
Reporting DNS Online after Pihole upgrade

**To Reproduce**
Steps to reproduce the behavior:

Upgrade Pihole:
  Pi-hole® v5.13, Web v5.16, FTL v5.18.2
  PADD v3.8.0   [✗] DNS is off!

Pihole is working fine

**Expected behavior**
padd.sh reporting the DNS is offline

**Screenshots**
[image]

**Additional context**

yubiuser, 2022-10-11:
See here https://github.com/pi-hole/PADD/issues/251#issuecomment-1274682675

yubiuser, 2022-10-12:
Fixed by: https://github.com/pi-hole/PADD/releases/tag/v3.8.1

closed 2022-10-12 by yubiuser