pi-hole/PADD #266: Reinstate methods CleanPrintf and CleanEcho? (commit 2b406c1)

MikeDeSantis, 2022-10-18:
**Describe the bug**
Commit 2b406c1: The removed methods CleanPrintf and CleanEcho take away the ability to centrally pad spacing for horizontal display offset to compensate on different installations.  Please consider putting them back.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from 3.8.0 to 3.8.1

**Expected behavior**
Central printing methods available to manipulate as needed.  

**Screenshots**
[image]

**Additional context**
For those who auto-upgrade PADD using crontab, a simple command could be run in 3.8.0 and prior to change the printer padding after the upgrade finished.
For my specific hardware, it was:
sed -i 's/echo "${ceol}$1"/echo "      ${ceol}$1"/g' ~/padd.sh
sed -i 's/printf "$@"/printf "      $@"/g' ~/padd.sh

This is no longer possible with the central printer functions removed.  Printing now starts offscreen to the left,  such as showing the logo as "ADD" instead of my corrective spacing to show "PADD".

MikeDeSantis, 2022-10-18:
A correction on the sed commands, the spacing was removed when posting.  
<pre>sed -i 's/echo "${ceol}$1"/echo "          ${ceol}$1"/g' ~/padd.sh</pre>
<pre>sed -i 's/printf "$@"/printf "          $@"/g' ~/padd.sh</pre>

yubiuser, 2022-10-19:
Both functions were removed because

> Both were introduced with https://github.com/pi-hole/PADD/pull/74. They solved the issue that after re-sizing the screen artifacts could remain. `CleanPrintf` did not serve any purpose, it was rather an intermediate state during this initial PR and was kept because it fitted the `CleanEcho` function well. The "cleaning" was done in the format string of the `printf` function with `\e[0K` already. `CleanEcho` was broken since https://github.com/pi-hole/PADD/pull/106, as the line needs clearing after the new output was written, not before.

From: https://github.com/pi-hole/PADD/pull/257

They were never intended for the
>  ability to centrally pad spacing for horizontal display offset to compensate on different installations.

What you did is a "hack" of the functions, they served merely as an anchor for your `sed` search string.

I won't re-introduce a function that does nothing, but I see some appeal in having a function that could move the whole `PADD` output "freely" on the available screen. For vertical positioning it's easy (only modify the `tput cup 0 0`), but for horizontal movement one needs to modify each line separately. 

____

Until something like this might be implemented you need to modify the relevant 26 lines of "mega" output by hand to add the desired amount of spaces..

yubiuser, 2022-10-24:
@MikeDeSantis 

I worked on this feature and implemented a way to set x/y offsets (Details: https://github.com/pi-hole/PADD/pull/268). It would be great if you could test the new version. You can find it here: https://github.com/pi-hole/PADD/blob/offset/padd.sh

To set an offset, just start PADD with `./padd.sh -xoff XX -yoff YY` (replace XX and YY with the appropriate number of lines/columns the Dashboard needs shifting).

MikeDeSantis, 2022-10-24:
I have tested the -xoff and -yoff implementation and it works exactly as expected.  This completely satisfies my need.  Many thanks @yubiuser!

closed 2022-10-24 by MikeDeSantis