pi-hole/PADD #273: Updating Pi-hole Core to 5.14.1 still shows "Red" in PADD

kagegoomba, 2022-11-15:
Updated my Pi-hole to latest version - PADD still says its outdated - already updated twice - updated PADD - no change.
Not sure if bug or other. Everything else is fine - working great - just annoyed by the version notice.

yubiuser, 2022-11-15:
Note: we should use the version file instead of using `pihole -v`

yubiuser, 2022-11-15:
Ok, that's strange. I don't see it in red - but I'll prepare something anymway.

yubiuser, 2022-11-15:
@kagegoomba 

Please check, if this version fixes the issue for you.

https://github.com/pi-hole/PADD/blob/versions_file/padd.sh

kagegoomba, 2022-11-15:
Don't suppose you could pinpoint want lines changed - don't exactly have an easy way to drop this file on my device short of pulling a new copy off the server ( I suspect this isn't official yet)

yubiuser, 2022-11-15:
You can see the changes here: https://github.com/pi-hole/PADD/pull/274/files

Easiest is probably to replace the whole `padd.sh` with the content from here: 
https://raw.githubusercontent.com/pi-hole/PADD/versions_file/padd.sh

mikegee81, 2022-11-15:
Hi @yubiuser I tried the updated padd.sh at the link you provided. Pi-hole v5.14.1, Web v5.17, and FTL v5.19.1 still display in red.

yubiuser, 2022-11-15:
What's the content of your `/etc/pihole/versions`?

mikegee81, 2022-11-15:
`CORE_BRANCH=master
WEB_BRANCH=master
FTL_BRANCH=master
CORE_VERSION=v5.14.1
WEB_VERSION=v5.17
FTL_VERSION=v5.19.1
GITHUB_CORE_VERSION=
GITHUB_WEB_VERSION=
GITHUB_FTL_VERSION=
CORE_HASH=21026d9
GITHUB_CORE_HASH=
WEB_HASH=da2764e
GITHUB_WEB_HASH=
FTL_HASH=b48b3e1f
GITHUB_FTL_HASH=
`

yubiuser, 2022-11-15:
It seems Pi-hole failed to get the latest version information from Github.
Please run `pihole updateckecker` and see if solved the issue.

P.S. This is not an issue with `PADD` but Pi-hole itself.

mikegee81, 2022-11-15:
Yes, that fixed it. Thanks so much!

kagegoomba, 2022-11-16:
No change - modded the file as you directed.
also tried pihole updatechecker after the changes.

kagegoomba, 2022-11-16:
`CORE_BRANCH=master
WEB_BRANCH=master
FTL_BRANCH=master
CORE_VERSION=v5.14.1
WEB_VERSION=v5.17
FTL_VERSION=v5.19.1
GITHUB_CORE_VERSION=v5.14.1
GITHUB_WEB_VERSION=v5.17
GITHUB_FTL_VERSION=v5.19.1
CORE_HASH=21026d94
GITHUB_CORE_HASH=21026d9
WEB_HASH=da2764e5
GITHUB_WEB_HASH=da2764e
FTL_HASH=b48b3e1f
GITHUB_FTL_HASH=b48b3e1
`

yubiuser, 2022-11-16:
This looks good. 

Please run `bash -x padd.sh`. This will produce **a lot** of output, you can cancel after 3 seconds. Please share the output.

kagegoomba, 2022-11-16:
```
+ LC_ALL=C
+ LC_NUMERIC=C
+ padd_version=v3.8.1
++ date +%s
+ LastCheckVersionInformation=1668634177
++ date +%s
+ LastCheckNetworkInformation=1668634177
++ date +%s
+ LastCheckSummaryInformation=1668634177
++ date +%s
+ LastCheckPiholeInformation=1668634177
++ date +%s
+ LastCheckSystemInformation=1668634177
++ nproc --all
+ core_count=4
++ printf '\033'
+ rSI='
   ed_text=''
+ green_text=''
+ yellow_text=''
+ blue_text=''
+ magenta_text=''
+ cyan_text=''
+ reset_text=''
+ clear_line=''
+ bold_text=''
+ blinking_text=''
+ dim_text=''
+ check_box_good='[✓]'
+ check_box_bad='[✗]'
+ check_box_question='[?]'
+ check_box_info='[i]'
+ pico_status_ok='[✓] Sys. OK'
+ pico_status_update='[i] Update'
+ pico_status_hot='[✗] Sys. Hot!'
+ pico_status_off='[✗] Offline'
+ pico_status_ftl_down='[i] FTL Down'
+ pico_status_dns_down='[✗] DNS Down'
+ pico_status_unknown='[?] Stat. Unk.'
+ mini_status_ok='[✓] System OK'
+ mini_status_update='[i] Update avail.'
+ mini_status_hot='[✗] System is hot!'
+ mini_status_off='[✗] Pi-hole off!'
+ mini_status_ftl_down='[i] FTL down!'
+ mini_status_dns_down='[✗] DNS off!'
+ mini_status_unknown='[?] Status unknown'
+ full_status_ok='[✓] System is healthy.'
+ full_status_update='[i] Updates are available.'
+ full_status_hot='[✗] System is hot!'
+ full_status_off='[✗] Pi-hole is offline'
+ full_status_ftl_down='[i] FTL is down!'
+ full_status_dns_down='[✗] DNS is off!'
+ full_status_unknown='[?] Status unknown!'
+ mega_status_ok='[✓] Your system is healthy.'
+ mega_status_update='[i] Updates are available.'
++ tput cols
+ console_width=80
++ tput lines
+ console_height=24
+ SizeChecker
+ '[' 80 -lt 20 ']'
+ '[' 24 -lt 10 ']'
+ '[' 80 -lt 24 ']'
+ '[' 24 -lt 12 ']'
+ '[' 80 -lt 30 ']'
+ '[' 24 -lt 16 ']'
+ '[' 80 -lt 40 ']'
+ '[' 24 -lt 18 ']'
+ '[' 80 -lt 53 ']'
[... 1125 lines cut ...]
SYSTEM ====================================================
+ printf ' %-10s%-39s\n' Uptime: '0 days, 21 hours, 25 minutes'
 Uptime:   0 days, 21 hours, 25 minutes
+ printf ' %-10s%-21s%-10s%-4s, %-4s, %-4s\n' 'CPU Temp:' $'40.4\302\260C' 'CPU
Load:' 1.08 0.84 0.82
 CPU Temp: 40.4°C              CPU Load: 1.08, 0.84, 0.82
+ printf ' %-10s[%-10s] %-6s %-10s[%-10s] %-5s' Memory: '··········' 2.2% 'CPU L
oad:' '■■■·······' 27.0%
 Memory:   [··········] 2.2%   CPU Load: [■■■·······] 27.0%+ tput ed
+ pico_status='[✓] Sys. OK'
+ mini_status='[✓] System OK'
+ tiny_status='[✓] System is healthy.'
+ full_status='[✓] System is healthy.'
+ mega_status='[✓] Your system is healthy.'
+ sleep 5
^C++ setterm -cursor on
++ echo

++ exit 0
+ setterm -cursor on
+ echo

+ exit 0
<redacted>@raspberrypi:~ $
```

kagegoomba, 2022-11-16:
[image]

kagegoomba, 2022-11-17:
Well this is awkward - it just turned green just now.
Didn't touch it - didn't do nothing to it since last mod of the .sh file from your instruction.
o.o;

yubiuser, 2022-11-17:
I'm glad it's solved now. I was a bit concerned already.

kagegoomba, 2022-11-17:
Only concerning thing is why/what caused it. Guess I'll report back when/if next update occurs if it repeats.

christianboyle, 2022-11-17:
Same issue here, but none of the above fixes have worked. I've tried the following:

I tried running `pihole updatechecker`, but it returns nothing and things are still red.

Also tried another `pihole up` and reboot this morning, but still no change.

Also tried replacing `padd.sh` with [https://raw.githubusercontent.com/pi-hole/PADD/versions_file/padd.sh](https://raw.githubusercontent.com/pi-hole/PADD/versions_file/padd.sh), but still nothing.

Anything else I might try?

yubiuser, 2022-11-17:
@christianboyle 

Only the `Pi-hole` version is in red? Does it also show "out-of-date" when it's starting up?

christianboyle, 2022-11-17:
@yubiuser Sorry, all 3 are red:

[image]

yubiuser, 2022-11-17:
I pushed an update to that branch. Please update your local version and try again. If it does not work, please provide the output of  `bash -x padd.sh`. The interesting lines look like

```
- Gathering version information...
+ GetVersionInformation mega
+ echo v5.14.1
+ grep -qE '^v[0-9]+([.][0-9]+){1,2}$'
++ VersionConverter v5.14.1
++ echo v5.14.1
++ tr -d '[:alpha:]'
++ awk -F. '{ printf("%d%03d%03d%03d\n", $1,$2,$3,$4); }'
+ core_version_converted=5014001000
++ VersionConverter v5.14.1
++ echo v5.14.1
++ tr -d '[:alpha:]'
++ awk -F. '{ printf("%d%03d%03d%03d\n", $1,$2,$3,$4); }'
+ core_version_latest_converted=5014001000
+ '[' 5014001000 -lt 5014001000 ']'
+ core_version_heatmap=''
+ '[' true = true ']'
+ echo v5.17-2-gf841f309
+ grep -qE '^v[0-9]+([.][0-9]+){1,2}$'
+ WEB_VERSION=N/A
+ web_version_heatmap=''
+ echo vDev-b72d3b7
+ grep -qE '^v[0-9]+([.][0-9]+){1,2}$'
+ FTL_VERSION=N/A
+ ftl_version_heatmap=''
++ curl --silent https://api.github.com/repos/pi-hole/PADD/releases/latest
++ grep '"tag_name":'
++ awk -F '"' '{print $4}'
+ padd_version_latest=v3.8.1
+ '[' -z v3.8.1 ']'
++ VersionConverter v3.8.1
++ echo v3.8.1
++ tr -d '[:alpha:]'
++ awk -F. '{ printf("%d%03d%03d%03d\n", $1,$2,$3,$4); }'
+ padd_version_latest_converted=3008001000
++ VersionConverter v3.8.1
++ echo v3.8.1
++ tr -d '[:alpha:]'
++ awk -F. '{ printf("%d%03d%03d%03d\n", $1,$2,$3,$4); }'
+ padd_version_converted=3008001000
+ '[' 3008001000 -lt 3008001000 ']'
+ padd_version_heatmap=''
+ '[' '' = true ']'
+ '[' '' = true ']'
+ version_status='Pi-hole is up-to-date!'

```

christianboyle, 2022-11-17:
@yubiuser Your change fixed it, all 3 are green now.

kagegoomba, 2022-11-17:
I'll do an update later tonight and see if anything happens - as I don't like a "modded" padd.sh off the main branch. :)

yubiuser, 2022-11-18:
> padd.sh file from the main master branch just now 

> except the Pi-hole core - its red 

This is expected, as

> We made some last minute changes before the last release which affect the way the version information are displayed. PADD now fails to get the right information.

You need to use the special version provided at 
https://raw.githubusercontent.com/pi-hole/PADD/versions_file/padd.sh

to fix the issue. Or you wait until this fix has been released with a new PADD version.

kagegoomba, 2022-11-18:
Ah - misread - ill try that again when I get a chance.

kagegoomba, 2022-11-19:
came up totally green and normal with the modded file. It's happy again. :)

kagegoomba, 2022-11-28:
FYI - update to pi-hole core today - current PADD File is working fine thusfar.

yubiuser, 2022-12-11:
Should be fixed now with release of https://github.com/pi-hole/PADD/releases/tag/v3.9.0

closed 2022-12-11 by yubiuser