pi-hole/PADD #275: 2.4" Waveshare screen doesn't work?

impy101, 2022-11-18:
I just got myself a compact 2.4" Waveshare screen to build into my case. However, it doesn't seem to be working with padd. All I get is a white screen.. I have configured it as suggested by the manufacturer but still nothing...

[https://m.media-amazon.com/images/I/71qLIN5hg9L._AC_SL1500_.jpg](url)

yubiuser, 2022-11-18:
`PADD` is a terminal application and will run on any display that can display a terminal. 
It looks like you have a hardware issue - we can't provide support for this. Please contact the manufacturer/hardware forum for guidance how to get the display working.

closed 2022-11-18 by yubiuser

impy101, 2022-11-18:
> `PADD` is a terminal application and will run on any display that can display a terminal. It looks like you have a hardware issue - we can't provide support for this. Please contact the manufacturer/hardware forum for guidance how to get the display working.

I have this screen: https://www.waveshare.com/wiki/2.4inch_LCD_Module

I already followed the tutorial for installing all the libraries etc and the test codes seem to work, but I can't seem to get padd running on the screen..

yubiuser, 2022-11-18:
So you already see a terminal or anything on the screen? Only PADD is not displayed?

rdwebdesign, 2022-11-18:
This looks like a hardware config problem and it is outside pi-hole devs reach.

You can try to get help on the product page:
- https://www.waveshare.com/wiki/2.4inch_LCD_Module#FAQ
- https://support.waveshare.com/hc/en-us/requests/new