pi-hole/PADD #276: Padd not showing on 5" DSI screen

impy101, 2022-11-25:
I have bought a brand new 5" Waveshare DSI screen. It seems to work properly. I have followed the PADD guide by setting PADD to auto run on startup etc. However, it's not doing that at all.

I can see the terminal etc just fine (see photo) but that's pretty much all it does.

[image]

yubiuser, 2022-11-25:
I can barley see anything on this blurry image. However, it looks like you some `*.service` failed. We don't have such a service. What did you use to install PADD/Pi-hole?

rdwebdesign, 2022-11-25:
I can't read too, but I think the service name is `pihole-oled.service` (or something like that).

Is this the service causing the problem?

Note:
`pihole-oled.service` is not an official Pi-hole service/script.

yubiuser, 2022-11-25:
I think you followed 
https://github.com/willdurand/pihole-oled

You need to ask the maintainer of this project for additional help.

closed 2022-11-25 by yubiuser

impy101, 2022-11-25:
I did a fresh install. Now padd FINALLY starts up. However, it's not utilizing the full screen

yubiuser, 2022-11-25:
This is expected. The maximum size it will use is 80 columns and 26 lines. There is a PR which will allow you to move the used area freely within your screen or automatically center the area. 
https://github.com/pi-hole/PADD/pull/268

impy101, 2022-11-25:
> 

Okay, so is there a guide on how I can apply that "PR"?

rdwebdesign, 2022-11-25:
Just replace the original file with the new one  (https://raw.githubusercontent.com/pi-hole/PADD/afb45427a12ee84ecc91021bce8f4dba4649d0b3/padd.sh)

rdwebdesign, 2022-11-25:
But this is still a Work in Progress.

It's not finished.

impy101, 2022-11-25:
> 

Yeah, doesnt work for me. It still stays the same size and now it flickers

rdwebdesign, 2022-11-25:
As I said: It's not finished.